turtlebot/turtlebot4 #431: Docking Station - Charger

CycleMark, 2024-06-21:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

N/A

### Type of issue

Battery

### Expected behaviour

I have the TB4 and TB2 - I know different bases/manufactures etc. 

Just wondered if you're able to provide details on how the charging circuity works? Reason I ask is that I would like to make a base that can change both. As is stands at the moment neither base will charge the other. 

TB4 base, for example, will switch on, then off the charging led when the TB2 is on it. Clearly it's detecting something's not right/as it expects. Would just like to know what.

Thanks

Mark

### Actual behaviour

Multi robot charging base.

### To Reproduce

N/A

RustyCPR, 2024-06-24:
Hi @CycleMark 
We did not make or manufacture the chargers or base units for TB2 and TB4. 

The Turtlebot 2 was built on a platform by http://kobuki.yujinrobot.com/ - who are no longer around it seems. 
The website is not reachable. I found this site that may have details: https://robots.ros.org/kobuki/
This model has been end of life/ support. 

For the TB4 is built on an iRobot Create 3 base unit and charger base - you could try posting the question here: https://github.com/iRobotEducation/create3_docs/discussions

Best regards,
Rusty

RustyCPR, 2024-07-08:
Closing due to non-activity

closed 2024-07-08 by RustyCPR