turtlebot/turtlebot4 #432: How to power off turtlebot 4

tedhuang96, 2024-06-24:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Simple Discovery

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

N/A

### Type of issue

Battery

### Expected behaviour

I thought there is a way to power off the robot by pressing the buttons.

### Actual behaviour

I did not find anywhere on the tutorial on how to power off the robot. I have to keep the robot off the deck to let the battery die. 

### To Reproduce

Power on, and then I did not find a way to power it off.

tedhuang96, 2024-06-25:
Not sure if this is the right way to do it, but I tried to run
```
ros2 service call /robot_power irobot_create_msgs/srv/RobotPower
```
and it returns
```
waiting for service to become available...
requester: making request: irobot_create_msgs.srv.RobotPower_Request()

response:
irobot_create_msgs.srv.RobotPower_Response(success=False, message='')
```

WallerJonathan, 2024-07-01:
To power off the robot, you have to hold down the power button for about 3 seconds. In addition, the TurtleBot has to be off the dock.

Sometimes the holding the power button doesn't work the first time, so you may have to try it multiple times; it should make a chime noise when it is turning off.

tedhuang96, 2024-07-03:
thanks! it works

closed 2024-07-03 by tedhuang96