turtlebot/turtlebot4 #435: Dummy names for gazebo plugin is valid?

AltinayGrass, 2024-07-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ros-humble-turtlebot4-simulator

### Type of issue

LIDAR

### Expected behaviour

scan topic and data

### Actual behaviour

yes topic but not logical data

### Error messages

```bash
No error message
```

### To Reproduce

`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true`

### Other notes

There are below line in "turtlebot4_description/urdf/sensors/rplidar.urdf.xacro"

```
  <gazebo reference="${name}_link">
    <xacro:ray_sensor sensor_name="${name}" gazebo="${gazebo}" 
                  update_rate="62.0" visualize="1" 
                  h_samples="640" h_res="1.0" h_min_angle="${-pi}" h_max_angle="${pi}" 
                  r_min="0.164" r_max="12.0" r_res="0.01">
                  <plugin name="dummy" filename="dummyfile"></plugin>
    </xacro:ray_sensor>
    <xacro:material_darkgray/>
  </gazebo>
```
Should we correct plugin name and filename?

RustyCPR, 2024-07-03:
Hi @AltinayGrass 
Can you please confirm the PC hardware you are running the simulation on? do you have a GPU? 
Please provide package versions, an example of data and/or error and screenshot from Gazebo including the visualization of the lidar rays.

Thank you,
Rusty

AltinayGrass, 2024-07-04:
Hi RustyCPR,

Thank you for your questions, I think that these are for solving my problem. However I still haven't got an answer to my question. Can you please confirm whether "dummy" means nonfunctional or need to be filled?

AltinayGrass, 2024-07-04:
[image]

RustyCPR, 2024-07-08:
@AltinayGrass dummy plugin is not used/non-functional input. 

The image you sent is how our TB4 expert expects the images to appear if you have no GPU or if the GPU is not being used correctly. If your PC does not have a GPU then they should try this: https://github.com/turtlebot/turtlebot4_simulator/issues/20#issuecomment-1302831141

Best regards,
Rusty

AltinayGrass, 2024-07-12:
Thanks for help RustyCPR, I check with "orge2" with CPU use, it's OK now. Even though I have the GPU ( NVIDIA Quadro P1000 with DELL precision 5530) not working with GPU use. How can I test GPU drivers, I configured it correctly?

closed 2024-07-12 by AltinayGrass

RustyCPR, 2024-07-22:
Hi @AltinayGrass 
You can test to ensure that the GPU is being used by launching the program that should be using the GPU and then running the nvidia-smi command. 
If the GPU is being used then it will show processes and GPU utilization. 
If the GPU is not being used then a common fix is to set the GPU to always be used (Open Nvidia X Server application, navigate to Prime Profiles and select NVIDIA (Performance Mode) . 
Apply these changes and reboot). This still requires you to use the correct NVIDIA driver.

The nvidia-smi command will display the driver version and cuda version for verification that it is using the expected driver and that cuda is properly setup.
Regards,
Rusty