turtlebot/turtlebot4 #437: Tilted robot base

tedhuang96, 2024-07-03:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

N/A

### Expected behaviour

The robot base is parallel to the ground.

### Actual behaviour

The robot base is tilted forward. The front part of the robot seems heavier than the part in the back, so the robot base is tilted forward when put on the ground. When we teleoperate the robot to move, there is a noticeable change of `/odom` in terms of z direction (~0.5 m change along z direction after moving forward several meters) caused by the tilted base. Attached are pictures of the turtlebot 4. We notice there is a hole in the front at the bottom of the base. Are we missing some parts that are supposed to support the front from tilting like rollers at the base bottom hole position? 

tilted base: https://drive.google.com/file/d/1vaXHZmcVTxf1FsyUMakplN-3oOU4RNZ9/view?usp=drive_link
base bottom: https://drive.google.com/file/d/1N9haWaZbMEL9DS8NCfq2MThHndZTtlGs/view?usp=drive_link
base bottom hole: https://drive.google.com/file/d/1dv1BFn5F82f9_1VG0CQrZ9eHexyFp_3g/view?usp=drive_link

### Error messages

```bash
N/A
```

### To Reproduce

N/A

RustyCPR, 2024-07-04:
Hi there @tedhuang96 
Your Create3 base is missing the front caster wheel. See photo here: https://iroboteducation.github.io/create3_docs/hw/data/bottom.jpg

You can purchase this caster wheel here: https://www.irobot.com/en_US/roomba-accessories/4624869.html

Best regards,
Rusty

tedhuang96, 2024-07-05:
Thanks! May I ask is the base supposed to come with that wheel or it is an extra part that needs to be purchased?

RustyCPR, 2024-07-05:
Hi @tedhuang96  The TB4s and Create3 do come with that wheel.

tedhuang96, 2024-07-05:
Thanks for the information! We don't think there is one in our package. We will contact customer service.

RustyCPR, 2024-07-06:
Hi @tedhuang96 where did you buy the TB4 from? 
We shipped that unit out in 2023.

tedhuang96, 2024-07-11:
We purchased the TB4 from RobotShop, and yes we indeed received the turtlebot4 in 2023. 

We tried to set up the robot back then, but the turtlebot4 documentation was not well organized at the moment, and we ran into many issues during the setup. We decided to wait until recently when the docs get much better to start the setup again.

We contacted RobotShop and they mentioned that the manufacturer typically handles the shipment of replacement parts directly. Does that mean we should contact Clearpath robotics to request the replacement parts with the contact below I found from Clearpath robotics official website?

[1 800 301 3863](tel:+1-800-301-3863)
[info@clearpathrobotics.com](mailto:info@clearpathrobotics.com)

RustyCPR, 2024-07-11:
Hi @tedhuang96 
Rusty here from the Clearpath Robotics Support team. 
Sorry to see you're experiencing this issue and not able to locate the missing caster wheel. 
Did you check the box the TB4 was stored in after you packed it up in 2023? 
Our team runs various tests on all units after building and setting them up. This type of non-conforming issue would be caught before packing/shipping the unit.  Not to mention iRobot ships us the Create3 base units fully tested. They also have great quality control. 

Due to the time that has elapsed since the initial 2023 shipment and parts like batteries and wheels not covered under product warranty due to being wear and tear items - there is not much I can do here. 

My initial recommendation was to source the caster wheel here: https://www.irobot.com/en_US/roomba-accessories/4624869.html

Kind regards,
Rusty

RustyCPR, 2024-07-22:
Closing this ticket - solution provided. 
 Feel free to reply if needed. 
Thanks,
Rusty

closed 2024-07-22 by RustyCPR