turtlebot/turtlebot4 #438: SLAM not working from PC - rplidar_link discarded

SamanthaSmith04, 2024-07-08:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

samubuntu@samubuntu:~$ dpkg -s ros-$ROS_DISTRO-turtlebot4-navigation
Package: ros-humble-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2470
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: arm64
Version: 1.0.4-1jammy.20240530.144553
Depends: ros-humble-nav2-bringup, ros-humble-nav2-simple-commander, ros-humble-slam-toolbox, ros-humble-ros-workspace
Description: Turtlebot4 Navigation

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

The updated map should appear in RViz along with LiDAR data when both `ros2 launch turtlebot4_navigation slam.launch.py` and `ros2 launch turtlebot4_viz view_robot.launch.py` are run on the PC.

I can get the map to work when running `ros2 launch turtlebot4_navigation slam.launch.py` through ssh on the raspberry pi and `ros2 launch turtlebot4_viz view_robot.launch.py` on the PC.

### Actual behaviour

A static image of the map is shown only, no lidar data is displayed, and most of the tf and robotmodel information is not connected.
[image]

Instead of registering the Lidar sensor, it just keeps printing that the message filter is dropping the rplidar_link frame

### Error messages

```bash
Running SLAM from PC
samubuntu@samubuntu:~$ ros2 launch turtlebot4_navigation slam.launch.py 
[INFO] [launch]: All log files can be found below /home/samubuntu/.ros/log/2024-07-08-11-16-16-363943-samubuntu-3634
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [3635]
[sync_slam_toolbox_node-1] [INFO] [1720451776.448386010] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1720451776.483089094] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1720451776.483291135] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1720451776.912736969] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866386.732 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1720451777.112955011] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866386.855 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1720451777.183319427] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866386.991 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1720451777.317997177] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866387.126 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1720451777.441654719] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866387.260 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1720451777.639544636] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866387.395 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1720451777.711557386] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866387.519 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1720451777.847515761] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866387.654 for reason 'discarding message because the queue is full'
[... 243 lines cut ...]
[sync_slam_toolbox_node-1] [INFO] [1719866482.300640369] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.029 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866482.434570447] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.165 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866482.557655437] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.301 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866482.692494655] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.435 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866482.827504741] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.558 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866482.962973915] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.693 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.098270924] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.828 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.220610126] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866482.963 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.355588453] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.099 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.490889166] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.221 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.625856382] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.356 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.761243446] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.492 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866483.896142903] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.626 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866484.019036692] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.762 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866484.154446126] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866483.896 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866484.289666599] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866484.019 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1719866484.424249430] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1719866484.155 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[image]

### Create 3 logs
Here is my Create3 log too:
[messages(1).txt](https://github.com/user-attachments/files/16131390/messages.1.txt)

Let me know if there is any other information that I should provide!

RustyCPR, 2024-08-09:
Hi @SamanthaSmith04, thanks for your patience, I was working with our Engineering team on your concerns, there are a few items we would like you to try:

1. Setup the NTP config on the Create3 so that the Create3 time syncs with the RPi instead of attempting to sync over the internet - see instructions in ticket #216

2. The default NTP Ubuntu time syncing systems (timedatectl and timesyncd) are insufficient when you need to offer time syncing services to other devices on the network.
See these instructions: https://ubuntu.com/server/docs/how-to-serve-the-network-time-protocol-with-chrony. Please follow the instructions up until and including the section that is "Enable serving the Network Time Protocol".
Specifically you'll need to install chrony and then "allow" the RPi access to the server . 
You also need to make sure that you restart chrony after making any changes to the file (sudo systemctl restart chrony.service). As an additional note make sure that this device has a static IP so that the RPi can always find it at a given IP address.

3. You'll need to redirect the RPi to sync with the new NTP server that you've set up. 
Open the /etc/chrony.conf file and add a new server line similar to the one you've used for the Create3 but changing the IP address to point at the NTP server set up in the previous step. 
This is what will tell the RPi to sync to your computer and then you can verify that after they have restarted chrony that you are able to sync properly.

Any questions please let me know, 
Rusty

RustyCPR, 2024-09-09:
Hi @SamanthaSmith04 any progress or updates?

RustyCPR, 2024-09-23:
Closing due to no activity.

closed 2024-09-23 by RustyCPR