turtlebot/turtlebot4 #439: Turtlebot4 gazebo simulator cmd_vel topic issue

dartik24, 2024-07-08:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

Not known

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

When running the turtlebot4 gazebo simulation for ros2 humble and publishing data to cmd_vel topic the robot should move accordingly within the simulation.

### Actual behaviour

When running the turtlebot4 gazebo simulation for ros2 humble there no reaction/very delayed reaction to published data in cmd_vel topic. 

### Error messages

```bash
N/A
```

### To Reproduce

Launch the simulation via the command below:
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true
Run any simple publisher to the cmd_vel topic.

smatarCPR, 2024-07-08:
Hello @dartik24 ,
Thank you for reaching out on the Turtlebot4 Github page. I am very sorry to hear that you are having issues with the simulator. Out of curiosity, did you press the play button found in the bottom left right corner?
[image]
Let me know if you have. Looking forward to hearing from you soon.
Best Regards,
Saif

dartik24, 2024-07-08:
Yes the play button is pressed, the robot briefly moves off the dock but then it does not respond to any further movement commands from either the keyboard or on screen commands.

smatarCPR, 2024-07-08:
Hello @dartik24 ,
Would it be possible to share some screenshots, and the terminal output from the terminal which launched Gazebo?

Looking forward to hearing from you soon.
Best Regards,
Saif

dartik24, 2024-07-08:
[image]
This issue repeats continuously after bootup.
Additionally it should be noted this is being ran in WSL2 however the same issue persists when running on a Linux native machine.

dartik24, 2024-07-09:
[image]

Upon further testing here are some additional errors which pop up after setting the robot into undocked mode.

closed 2024-07-15 by dartik24

aharshac, 2024-07-16:
@dartik24 How did you solve this?

dartik24, 2024-07-16:
@aharshac This issue is not fully solved, I closed it and re opened it here https://github.com/turtlebot/turtlebot4/issues/441 for a more concise summary.