turtlebot/turtlebot4 #440: Turtlebot4 Robot Unable to Connect to MindRove Armband's WIFI

justinlxiang, 2024-07-11:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

N/A

### Type of issue

Networking

### Expected behaviour

Should be able to connect to the MindRove WIFI.

### Actual behaviour

Jx372Jul 9, 2024, 13:13 EDTHello Clearpath Robotics Support Team,I am experiencing an issue with my Turtlebot4 robot. It is unable to connect to the WIFI network provided by my MindRove Armband (https://mindrove.com/armband/?gad_source=1&gclid=Cj0KCQjwv7O0BhDwARIsAC0sjWPQB2PcqAwBx6SwYY3h2O7ifPX8P3XsVON1-0GGFCLWZJLhBDH-ToYaAoYXEALw_wcB).I have verified that the WIFI network from the armband is active and other devices can connect to it without any issues. 
However, the Turtlebot4 fails to establish a connection when using the turtlebot4-setup command or the network manager command line interface, despite seeing the armband’s SSID in a list of detectable WIFIs.

Here are some details that might help diagnose the issue:
- The WIFI network SSID and password have been double-checked and are correct.
- Attempted to use an external WIFI adapter attached to the Turtlebot4’s Raspberry Pi and it still wouldn’t connect.
- The Turtlebot4 has the latest firmware and software updates installed.
- Other WIFI networks can be connected to by the Turtlebot4 without any problems.

The MindRove Armband collects EMG and IMU data, which I need to integrate with the Turtlebot4 for my project. I would appreciate any guidance or support you can provide to resolve this issue.
Thank you.

### To Reproduce

Can't really reproduce as you probably don't have this specific armband.

smatarCPR, 2024-07-22:
Hello @justinlxiang ,
Thank you for reaching out on the Turtlebot4 github page. I am very sorry that you are having issues with one of our products. Looking into the MindRove Armband, from our research it appears that the Armband may not be using the standard settings that the Turtlebot4 tool uses.

Would it be possible to verify if in the user manual there are instructions for connecting the unit to a device running Linux using netplan? Alternatively you may also attempt contacting MindRove themselves https://mindrove.com/contact/

Once these details have been determined, create a custom netplan yaml file in /etc/netplan with the details shared by MindRove.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

RustyCPR, 2024-08-26:
Closing due to no reply / activity on this ticket.

closed 2024-08-26 by RustyCPR