turtlebot/turtlebot4 #441: Turtlebot 4 simulation ignores publishing to cmd_vel.

dartik24, 2024-07-15:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Windows / MAC

### Built from source or installed?

Installed

### Package version

N/A

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

When running teleop keyboard or utilizing on screen gazebo simulation controls the expected behavior is for the robot to move accordingly. 

### Actual behaviour

[image]
The above error pops up whenever publishing to cmd_vel is attempted an the robot briefly stutters after being undocked but does not move.

### Error messages

```bash
[motion_control-35] [WARN] [1720543988.314924202] [motion_control]: Ignoring Velocities command while an autonomous behavior is running!
```

### To Reproduce

Launch the simulation via ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true
 command then press the play button and set the robot into undocked mode. Following this try to move the robot using the build in control panel.

smatarCPR, 2024-07-22:
Hello @dartik24 
Thank you for reaching out on the Turtlebot 4 Githu. I am very sorry to hear that you are having issues with one of our toolkits. From the details provided it appears that the simulated Turtlebot is choosing to ignore teleop commands, due to the autonomy running (you have launched the simulator with slam and nav2 enabled). If you attempt to launch the simulator, with only RVIZ running, are you able to control the Turtlebot using the control panel?

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-08-26 by RustyCPR

hanzheteng, 2025-03-19:
Hello @RustyCPR @smatarCPR , I recently run into a similar issue that it ignores the teleop commands due to autonomous behavior is running. I have tried the recommendations above that launching turtlebot4 robot with slam and nav2 disabled, but it did not help. Is there any way that we can disable this autonomous behavior completely or set cmd_vel to have higher priority? We want to use our script to control the robot freely via cmd_vel.

In my case, the ultimate goal is to remove the docking station, and move the robot freely around the origin area. However, after removing the dock, whenever the robot gets closer to where the dock originally located, it will launch some autonomous behavior that presents it from moving forward.