turtlebot/turtlebot4 #442: Camera not consistently working

xylhal, 2024-07-16:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Not sure what version it is

### Type of issue

Camera

### Expected behaviour

OAKD Camera to consistently publish images

### Actual behaviour

I observed multiple issues:

1. The `/oakd` topics are not always consistently present when list the topics
2. The most common behavior I have observed is that the topic is present, but when I visualize it in Rviz, the preview just shows a blank image (despite the Turtlebot being undocked)
3. Sometimes, when I restart Rviz and the turtlebot, there is initial image that is shown in the Rviz, but it freezes at the same image even if I move the Turtlebot
4. When I attempt to echo the topic `/oakd/rgb/preview/image_raw`, I get the warning that "A message was lost !!!"

### To Reproduce

1. Undock the turtlebot
2. `ros2 launch turtlebot4_viz view_robot.launch.py`

RustyCPR, 2024-07-22:
Hi there @xylhal  
Thanks for reaching out. 

"Sometimes, when I restart Rviz and the turtlebot, " after rebooting are you sshing in again and relaunching RVIZ? 
Can you please provide the terminal outputs with any errors? 

Thank you,
Rusty

xylhal, 2024-07-23:
Hi there @RustyCPR 

I'm actually not launching rviz from the RPi. I am running everything remotely from another machine. So, I didn't do any ssh. What I meant by restarting rviz and turtlebot simply means physically restarting the turtlebot and creating a new terminal on my remote machine to start rviz.

I've attached 5 images below. 
1. Screenshot from view_robot.launch.py
[image]
2. Screenshot from Rviz when view_robot.launch.py is running in the terminal
[image]
3. Screenshot from view_model.launch.py
[image]
4. Screenshot from Rviz when view_model.launch.py is running in the terminal
[image]
5. Terminal when I echo the camera topic
[image]

Please let me know how else can I debug this?

xylhal, 2024-07-23:
I would add that I realize sometimes the terminal does print some messages when echoing the topic, but the Rviz remains blank.
[image]

RustyCPR, 2024-08-09:
Hi @xylhal 

Run: `ros2 launch turtlebot4_viz view_model.launch.py` if your not using mapping

Point 5 image with error "`A message was lost!!!`" What type of wifi network are you running? 
If you're on a 2.4GHz network can you try a faster 5GHz?  

Regards,
Rusty

xylhal, 2024-09-05:
Hi @RustyCPR 

I was using 5 GHz all along, but the problem has mainly gone away, though I'm not sure why. I'm facing another issue now, which I will open in a separate issue.

closed 2024-09-05 by xylhal