turtlebot/turtlebot4 #443: Turtlebot4 does not move in Ignition Fortress

aharshac, 2024-07-16:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

Installed by running `sudo apt install ros-humble-turtlebot4-simulator` which also installed the following dependencies:

Package|Version
-- | --
ros-humble-ackermann-steering-controller | 2.35.0-1jammy.20240611.150636
ros-humble-admittance-controller | 2.35.0-1jammy.20240611.152045
ros-humble-bicycle-steering-controller | 2.35.0-1jammy.20240611.150635
ros-humble-effort-controllers | 2.35.0-1jammy.20240611.150523
ros-humble-force-torque-sensor-broadcaster | 2.35.0-1jammy.20240611.150216
ros-humble-forward-command-controller | 2.35.0-1jammy.20240611.150246
ros-humble-ign-ros2-control | 0.7.9-1jammy.20240702.133731
ros-humble-imu-sensor-broadcaster | 2.35.0-1jammy.20240611.150756
ros-humble-irobot-create-common-bringup | 2.1.0-1jammy.20240611.152747
ros-humble-irobot-create-control | 2.1.0-1jammy.20240611.152548
ros-humble-irobot-create-ignition-bringup | 2.1.0-1jammy.20240702.134602
ros-humble-irobot-create-ignition-toolbox | 2.1.0-1jammy.20240611.143003
ros-humble-irobot-create-nodes | 2.1.0-1jammy.20240611.143008
ros-humble-position-controllers | 2.35.0-1jammy.20240611.150535
ros-humble-range-sensor-broadcaster | 2.35.0-1jammy.20240611.145853
ros-humble-ros2-controllers | 2.35.0-1jammy.20240611.152436
ros-humble-steering-controllers-library | 2.35.0-1jammy.20240611.150247
ros-humble-tricycle-controller | 2.35.0-1jammy.20240611.145926
ros-humble-tricycle-steering-controller | 2.35.0-1jammy.20240611.150639
ros-humble-turtlebot4-ignition-bringup | 1.0.2-1jammy.20240702.134808
ros-humble-turtlebot4-simulator | 1.0.2-1jammy.20240702.135349
ros-humble-velocity-controllers | 2.35.0-1jammy.20240611.150527

Ended up afterwards with:
[image]

### Type of issue

Select One

### Expected behaviour

The simulator is started with `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`.

### Actual behaviour

The robot oscillates in place on the dock for a moment after pressing the start simulation button. 

Undock function does not work:

> [motion_control-35] [INFO] [1721156313.253158752] [motion_control]: Received new undock goal
> [turtlebot4_node-30] [ERROR] [1721156313.253800347] [turtlebot4_node]: undock goal was rejected by server
> [turtlebot4_node-30] [ERROR] [1721156313.253949995] [turtlebot4_node.rclcpp_action]: unknown goal response, ignoring...
> [turtlebot4_node-30] [ERROR] [1721156313.254056785] [turtlebot4_node.rclcpp_action]: unknown goal response, ignoring...
[... 436 lines cut ...]
[motion_control-35] [INFO] [1721157531.636356844] [motion_control]: E-Stop request: Set system E-Stop off, enabling motor power
[turtlebot4_node-30] [ERROR] [1721157535.682659323] [turtlebot4_node]: Service oakd/stop_camera unavailable.
[turtlebot4_node-30] [ERROR] [1721157535.693298306] [turtlebot4_node]: Service stop_motor unavailable.
[turtlebot4_node-30] [ERROR] [1721157545.240502696] [turtlebot4_node]: Power OFF
[turtlebot4_node-30] [INFO] [1721157546.241072663] [turtlebot4_node]: robot_power service available, sending request
[motion_control-35] [ERROR] [1721157546.241380744] [motion_control]: Power off request: Set system power off failed (not supported in sim)
[turtlebot4_node-30] [INFO] [1721157546.241802533] [turtlebot4_node]: robot_power service got results: Failed
[turtlebot4_node-30] [INFO] [1721157546.241876548] [turtlebot4_node]: Set system power off failed (not supported in sim)
[turtlebot4_node-30] [ERROR] [1721157551.290799167] [turtlebot4_node]: Power OFF
[turtlebot4_node-30] [INFO] [1721157552.291417463] [turtlebot4_node]: robot_power service available, sending request
[motion_control-35] [ERROR] [1721157552.292213818] [motion_control]: Power off request: Set system power off failed (not supported in sim)
[turtlebot4_node-30] [INFO] [1721157552.293008769] [turtlebot4_node]: robot_power service got results: Failed
[turtlebot4_node-30] [INFO] [1721157552.293191465] [turtlebot4_node]: Set system power off failed (not supported in sim)
[turtlebot4_node-30] [INFO] [1721157558.820561726] [turtlebot4_node]: Undocking
[turtlebot4_node-30] [INFO] [1721157558.820631063] [turtlebot4_node]: Waiting for undock action server
[turtlebot4_node-30] [INFO] [1721157559.821180905] [turtlebot4_node]: undock action server available, sending goal
[turtlebot4_node-30] [ERROR] [1721157559.821843092] [turtlebot4_node]: undock goal was rejected by server
[turtlebot4_node-30] [ERROR] [1721157559.822062458] [turtlebot4_node.rclcpp_action]: unknown goal response, ignoring...
[turtlebot4_node-30] [ERROR] [1721157559.822144260] [turtlebot4_node.rclcpp_action]: unknown goal response, ignoring...
[motion_control-35] [INFO] [1721157559.821961064] [motion_control]: Received new undock goal
```

### To Reproduce

`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`

aharshac, 2024-07-16:
Full log: [tb4_sim_error.log](https://github.com/user-attachments/files/16255693/tb4_sim_error.log)

RustyCPR, 2024-07-23:
Hi @aharshac 
Can you please detail the hardware specs of your PC you are running this on? 
It seems like the system does not have the required resources (CPU, RAM etc.) to run this simulation.
Best regards,
Rusty

aharshac, 2024-07-23:
Hello @RustyCPR 

I've seen this issue on two PCs. One with 16 GB RAM, 4GB 1050Ti and another with 32 GB RAM, 4 GB P2000. Both of them were running local Fast DDS discovery servers. This problem exists with *Create3 Ignition* as well, but not with *Gazebo Classic*.

I did not see this problem when I switched to Cyclone DDS with simple discovery. I want to use Turtlebot4 sim with multiple robots for frontier exploration, and I was unable to spawn more than one robot with Cyclone.

I think this issue is caused by *ros2_control* https://github.com/ros-controls/ros2_control/issues/1200. There's a fix https://github.com/ros-controls/ros2_control/pull/1562, but I was unable to build the branch locally.

Here's how I source my env
https://gist.github.com/aharshac/b07929459d00935bf6b0259144202e58 

Thanks

aharshac, 2024-07-23:
@RustyCPR Can you confirm if I'm sourcing my environment properly?
This is how I start the discovery serer
```sh
#!/bin/bash
source /opt/ros/humble/setup.bash
fastdds discovery -i 0 -p 11811
```

aharshac, 2024-07-23:
https://github.com/iRobotEducation/create3_sim/issues/194#issuecomment-2245957234

destogl, 2024-08-16:
> Hello @RustyCPR
> 
> I've seen this issue on two PCs. One with 16 GB RAM, 4GB 1050Ti and another with 32 GB RAM, 4 GB P2000. Both of them were running local Fast DDS discovery servers. This problem exists with _Create3 Ignition_ as well, but not with _Gazebo Classic_.
> 
> I did not see this problem when I switched to Cyclone DDS with simple discovery. I want to use Turtlebot4 sim with multiple robots for frontier exploration, and I was unable to spawn more than one robot with Cyclone.
> 
> I think this issue is caused by _ros2_control_ [ros-controls/ros2_control#1200](https://github.com/ros-controls/ros2_control/issues/1200). There's a fix [ros-controls/ros2_control#1562](https://github.com/ros-controls/ros2_control/pull/1562), but I was unable to build the branch locally.
> 
> Here's how I source my env https://gist.github.com/aharshac/b07929459d00935bf6b0259144202e58
> 
> Thanks

We have fixed the issues in all supported distros. Currently in source, but feel free to test and give us feedback!

RustyCPR, 2024-08-19:
Hi @aharshac 

Please try removing/unsetting the FASTRTPS_DEFAULT_PROFILES_FILE from your sourcing script. 
That profile is just pointing it towards the server you started and setting it as super client which is a duplication of the other environment variables that you have set.

aharshac, 2024-08-26:
@RustyCPR The simulator works as expected after https://github.com/ros-controls/ros2_control/pull/1562 was merged. Thanks @destogl for the fix.

closed 2024-08-26 by aharshac