turtlebot/turtlebot4 #444: Connecting Pi to passwordless network

Leo-Gut28, 2024-07-16:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.0-1jammy.20240530.114115 (ros-humble-turtlebot4-desktop)

### Type of issue

Networking

### Expected behaviour

I am following the [basic setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html) and I am trying to connect the Raspberry Pi to a network that doesn't have a password. It should just display the new IP on the screen.

### Actual behaviour

I followed the instructions but instead of showing me the new IP, it says "UNKNOWN", so I'm unable to SSH into it (I have to do it through ethernet). I double checked that it's the correct network name and I just left the password blank (just pressed enter).

### To Reproduce

Follow the instructions in the [setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html). But for the password, I just left it blank (pressed enter).

RustyCPR, 2024-07-24:
Hi @Leo-Gut28 
A WiFi password is required. Netplan the network config tool requires it. 
Rusty

RustyCPR, 2024-08-09:
Closed due to no activity

closed 2024-08-09 by RustyCPR