turtlebot/turtlebot4 #445: Nav2 Path Issue

sammahmeterdogan, 2024-07-22:
### Robot Model

Select One

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

`amcl:
  ros__parameters:
    use_sim_time: false
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 0.5
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: false
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
[... 650 lines cut ...]
`
My nav2_params.yaml 

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Hi, I'm using ROS 2 Humble and I'm mapping with the Slamtec S2 Lidar. I want to use a path in the map I generated, but whenever I run ros2 launch nav2_bringup navigation_launch.py, I keep getting the following error spam.
[image]

### Actual behaviour

ros2 launch nav2_bringup navigation_launch.py

### Error messages

```bash
[planner_server-3] [INFO] [1721648232.173436474] [global_costmap.global_costmap]: Message Filter dropping message: frame 'laser' at time 1721648231.773 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
```

### To Reproduce

-

sammahmeterdogan, 2024-07-22:
`amcl:
  ros__parameters:
    use_sim_time: false
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 0.5
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: false
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
    plugin_lib_names:
      - nav2_compute_path_to_pose_action_bt_node
      - nav2_compute_path_through_poses_action_bt_node
      - nav2_smooth_path_action_bt_node
      - nav2_follow_path_action_bt_node
      - nav2_spin_action_bt_node
      - nav2_wait_action_bt_node
      - nav2_assisted_teleop_action_bt_node
      - nav2_back_up_action_bt_node
      - nav2_drive_on_heading_bt_node
      - nav2_clear_costmap_service_bt_node
      - nav2_is_stuck_condition_bt_node
      - nav2_goal_reached_condition_bt_node
      - nav2_goal_updated_condition_bt_node
      - nav2_globally_updated_goal_condition_bt_node
      - nav2_is_path_valid_condition_bt_node
      - nav2_initial_pose_received_condition_bt_node
[... 255 lines cut ...]
  ros__parameters:
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.26, 0.0, 1.0]
    min_velocity: [-0.26, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
`

smatarCPR, 2024-08-01:
Hello @sammahmeterdogan ,
Thank you for reaching out on the Turtlebot 4 Github. I am very sorry to hear that you are having issues with one of our products.

From the error code supplied in terminal, it appears that there is a time sync issue between the raspberry pi and Create3 base unit. Have you made any modifications to the time syncing on the Create3?

One recommendation would be to do the following:
1. Setup NTP config Create3 to time sync Create3 and RPi - see instructions in [ticket #216](https://github.com/turtlebot/turtlebot4/issues/216).
2. Setup NTP server on your remote PC on the network - see instructions [here](https://ubuntu.com/server/docs/use-timedatectl-and-timesyncd)
3. Set up the RPi to sync to that PC
Run the command: systemctl status systemd-timesyncd and see when the timesync last occurred.
For the Raspberry Pi, you can use systemctl status chrony to see when the last sync was and you can force a sync by restarting the chrony service.
Ensure that the Raspberry Pi can ping the Ubuntu NTP servers using ping [ntp.ubuntu.com](http://ntp.ubuntu.com/)

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

RustyCPR, 2024-08-26:
Closing due to no updates or replies.

closed 2024-08-26 by RustyCPR