turtlebot/turtlebot4 #447: Pi not properly receiving battery or data from Create3, ros2 topic list empty

tharukakk1, 2024-07-23:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Select One

### Package version

ROS2 Humble

### Type of issue

Networking

### Expected behaviour

- all topics should appear in Pi ssh terminal when I run $ros2 topic list
- all topics should appear in desktop computer terminal when I run $ros2 topic list (ros domain id is set to 0, matching the robot)
- onboard screen should show correct battery level, btn 1-4 buttons should function and control the robot
- all of the above functionality should continue after reboot

In short, I'm trying to set up Simple Discovery on the robot. The ros2 topics are going missing from the Pi when I try to achieve this.

### Actual behaviour

- $ros2 topic list in Pi's terminal only shows /parameter_events and /rosout
- desktop's terminal $ros2 topic list also only shows /parameter_events & /rosout, even when the Pi showed the full list
- onboard screen shows 0% battery even though Create3 webserver shows 98%. buttons don't control screen, nor does the undock button get the robot to move.
- the first and the third point worked as intended yesterday. As part of setting up Simple Discovery, I rebooted the robot. it stopped working after reboot. Robot has been charging continuously overnight. Tried rebooting several times.
- I have two standard tb4's which is why i'm trying to set up simple discovery. If Discovery Server is the better method, please let me know. The second tb4 is not set up on the network yet.

### To Reproduce

1. turn on tb4 using dock
2. wait for full start up (~30 mins to be safe)
3. confirm functionality using light ring as an indicator
4. open create3 webserver, confirm all settings are there, and battery state in About
5. open ssh terminal $ssh ubuntu@[ip address]
6. run $ros2 topic list in Pi terminal
7. run $ros2

tharukakk1, 2024-07-23:
Update: The onboard screen fully works; I'm even able to use the "btn" buttons to undock and dock the robot. All of the topics show up in the Pi's terminal and they all have the namespace I assigned.

What's Not Working:
- running ROS tests in Pi's terminal
- running teleop commands in Pi's terminal
- running teleop commands in desktop's terminal

RustyCPR, 2024-08-26:
Hi there @tharukakk1 
We recommend using [Discovery Server ](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#discovery-server)

Let us know if you have any concerns or questions,
Rusty

closed 2024-09-23 by smatarCPR