turtlebot/turtlebot4 #451: Oak-D pro stereo depth and other possible sensors of the camera

AncientGrey, 2024-07-31:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

ros-humble-turtlebot4-turtlebot4_robot

### Type of issue

Camera

### Expected behaviour

Hi, is it possible to provide a camera (oak-d pro) config .yaml file, where every Sensor and Actuator is included and activated?
That way it is easier to get an overview of all the possible uses of the camera and the turtlebot.

### Actual behaviour

I have been trying to get the depth image to work, but i have troubles with the camera config.

### To Reproduce

edit oakd_pro.yaml
run oakd_bringup

RustyCPR, 2024-08-09:
Hi @AncientGrey 
Here are some example launch files: https://github.com/luxonis/depthai-ros/tree/humble/depthai_examples/launch

Please see this ticket reply related to additional depth camera topics:  https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838

Regards,
Rusty

AncientGrey, 2024-08-12:
would it be possible to compile all options in one .yaml file, with all options for the oakd_pro.yaml?

RustyCPR, 2024-08-19:
Hi @AncientGrey 
Not clear what your request is. Can you please provide a more in-depth description of what you are looking for?

AncientGrey, 2024-09-09:
I was looking for a full config for the Oak-D Pro camera. If i change the pipeline type to anything apart from 'RGB' the Launch fails with an Error. 
I would like to use the depth data of the camera to approach objects.
Following are my config paranmeters of the oakd:pro.yaml file and the Error of the launch.

```
/oakd:
  ros__parameters:
    camera:
      i_calibration_dump: false
      i_enable_imu: true
      i_enable_ir: true
      i_external_calibration_path: ''
      i_floodlight_brightness: 400
      i_ip: ''
      i_laser_dot_brightness: 800
      i_mx_id: ''
      i_nn_type: none
      i_pipeline_dump: true
      i_pipeline_type: RGBD
      i_publish_tf_from_calibration: false
      i_restart_on_diagnostics_error: false
      # i_tf_base_frame: oak
      # i_tf_cam_pitch: '0.0'
      # i_tf_cam_pos_x: '0.0'
      # i_tf_cam_pos_y: '0.0'
      # i_tf_cam_pos_z: '0.0'
      # i_tf_cam_roll: '0.0'
      # i_tf_cam_yaw: '0.0'
      # i_tf_camera_model: ''
      # i_tf_camera_name: oak
      # i_tf_custom_urdf_location: ''
      # i_tf_custom_xacro_args: ''
      i_tf_imu_from_descr: 'true'
      # i_tf_parent_frame: oak-d-base-frame
      # i_usb_port_id: ''
      i_usb_speed: SUPER_PLUS
    rgb:
      i_add_exposure_offset: false
      i_board_socket_id: 0
      i_calibration_file: ''
      i_disable_node: false
      i_enable_feature_tracker: false
      i_enable_lazy_publisher: true
      i_enable_nn: false
      i_enable_preview: false
      i_exposure_offset: 0
      i_fps: 30.0
      i_fsync_continuous: false
      i_fsync_trigger: false
      i_get_base_device_timestamp: false
      i_height: 720
      i_interleaved: false
      i_isp_den: 3
      i_isp_num: 2
      i_keep_preview_aspect_ratio: true
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_max_q_size: 30
      i_output_isp: true
      i_preview_height: 300
      i_preview_size: 300
      i_preview_width: 300
      i_publish_topic: true
      i_resolution: 1080P
      i_reverse_stereo_socket_order: false
      i_sensor_img_orientation: AUTO
      i_set_isp3a_fps: false
      i_set_isp_scale: true
      i_simulate_from_topic: false
      i_simulated_topic_name: ''
[... 126 lines cut ...]
      i_right_rect_low_bandwidth: false
      i_right_rect_low_bandwidth_quality: 50
      i_right_socket_id: 2
      i_set_disparity_to_depth_use_spec_translation: false
      i_set_input_size: false
      i_socket_name: rgb
      i_spatial_nn_source: right
      i_stereo_conf_threshold: 240
      i_subpixel: false
      i_update_ros_base_time_on_ros_msg: false
      i_width: 1280
    use_sim_time: false
```
```
Nov 21 21:00:09 ubuntu turtlebot4-start[1112]: [joint_state_publisher-8] [INFO] [1700600409.710459042] [blastoise.joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
Nov 21 21:00:12 ubuntu turtlebot4-start[1112]: [component_container-6] [INFO] [1700600412.662308717] [blastoise.oakd]: Camera with MXID: 18443010B162BC0800 and Name: 1.1.2 connected!
Nov 21 21:00:12 ubuntu turtlebot4-start[1112]: [component_container-6] [INFO] [1700600412.663868408] [blastoise.oakd]: USB SPEED: SUPER
Nov 21 21:00:12 ubuntu turtlebot4-start[1112]: [component_container-6] [INFO] [1700600412.695560891] [blastoise.oakd]: Device type: OAK-D-PRO
Nov 21 21:00:12 ubuntu turtlebot4-start[1112]: [component_container-6] [INFO] [1700600412.700948703] [blastoise.oakd]: Pipeline type: RGBD
Nov 21 21:00:13 ubuntu turtlebot4-start[1112]: [component_container-6] [INFO] [1700600413.174712718] [blastoise.oakd]: Finished setting up pipeline.
Nov 21 21:00:13 ubuntu turtlebot4-start[1112]: [component_container-6] [INFO] [1700600413.178229506] [blastoise.oakd]: Saving pipeline schema to: /tmp/18443010B162BC0800_pipeline.json
Nov 21 21:00:17 ubuntu turtlebot4-start[1112]: [component_container-6] [18443010B162BC0800] [1.1.2] [1700600417.479] [host] [warning] Device crashed, but no crash dump could be extracted.
Nov 21 21:00:19 ubuntu turtlebot4-start[1112]: [component_container-6] [ERROR] [1700600419.071206213] [blastoise.oakd_container]: Component constructor threw an exception: Device already closed or disconnected: Input/output error
Nov 21 21:00:19 ubuntu turtlebot4-start[1112]: [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oakd' of type 'depthai_ros_driver::Camera' in container '/blastoise/oakd_container': Component constructor threw an exception: Device already closed or disc>
```

RustyCPR, 2025-08-13:
Hi there @FGJoseMiguel-AIST  - your best bet is post your question here: https://github.com/luxonis/depthai-ros

I'll go ahead and close this ticket. 
Thanks,
Rusty

closed 2025-08-13 by RustyCPR