turtlebot/turtlebot4 #454: Action doesn't work

taara67, 2024-08-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

H2.6
fastdds (default)
usb

### Type of issue

Other

### Expected behaviour

I want to use the ros2 action to perform the create3 operation.

First, I executed the following command on the command line

ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance "{distance: 0.5,max_translation_speed: 0.15}"

### Actual behaviour

It didn't work.
I got a few errors about the message size,
but it was basically just waiting.
I've seen delays of a few seconds, but even after waiting 10 minutes it still didn't work.

Additionally, a happy sound suddenly started. The cause is unknown, but it appears to reboot.

### Error messages

```bash
^C
ubuntu@ubuntu:~$ ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance "{distance: 0.5,max_translation_speed: 0.15}"
Waiting for an action server to become available...
Sending goal:
     distance: 0.5
max_translation_speed: 0.15

2024-08-02 08:41:52.524 [RTPS_READER_HISTORY Error] Change payload size of '32' bytes is larger than the history payload size of '19' bytes and cannot be resized. -> Function can_change_be_added_nts
            

^C
ubuntu@ubuntu:~$ ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance "{distance: 0.5,max_translation_speed: 0.15}"
Waiting for an action server to become available...
Sending goal:
     distance: 0.5
max_translation_speed: 0.15

^C
```

### To Reproduce

I flashed the humble image for tb4 and updated it to the latest version along with the C3 firmware. I used the setup tool.
After that, all you have to do is run the command.
The list of topics and actions is as follows.

ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/rgb/preview/image_raw
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
ubuntu@ubuntu:~$ ros2 node list
/_internal/composite_hazard
/_internal/kinematics_engine
/_internal/mobility
/_internal/stasis
/analyzers
/joint_state_publisher
/joy_linux_node
/launch_ros_1217
/mobility_monitor
/motion_control
/oakd
/oakd_container
/robot_state
/robot_state_publisher
/static_transform
/system_health
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
/ui_mgr
ubuntu@ubuntu:~$ ros2 action list
/dock
/drive_distance
/led_animation
/undock
/wall_follow

dgel27, 2024-08-02:
Hi,
I think you need to install create3 messages package.
check if it installed:
apt list --installed | grep irobot
if you cannot see ros-humble-irobot-create-msgs, install it:
sudo apt install -y ros-humble-irobot-create-msgs

br,

taara67, 2024-08-05:
thank you for your reply
-----
ubuntu@ubuntu:~/turtlebot4_ws$ apt list --installed |grep irobot

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-irobot-create-control/jammy,now 2.1.0-1jammy.20240801.165108 arm64 [installed]
ros-humble-irobot-create-description/jammy,now 2.1.0-1jammy.20240729.033517 arm64 [installed,automatic]
ros-humble-irobot-create-msgs/jammy,now 2.1.0-1jammy.20240730.201125 arm64 [installed]
------

it was already installed in Raspi 4

taara67, 2024-08-05:
The send goal was executed!
```ubuntu@ubuntu:~/turtlebot4_ws$ time ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance "distance: 0.3
max_translation_speed: 2"
Waiting for an action server to become available...
Sending goal:
     distance: 0.3
max_translation_speed: 2.0

Goal accepted with ID: 161fe707c3c047168555455001fad3c4

Result:
    pose:
  header:
    stamp:
      sec: 1671295893
      nanosec: 515621639
    frame_id: odom
  pose:
    position:
      x: 0.137577086687088
      y: 1.928069829940796
      z: 0.011613388545811176
    orientation:
      x: -0.002767911152024286
      y: -0.004743525546562102
      z: 0.6954821418999861
      w: 0.7185223920877143

Goal finished with status: SUCCEEDED

real	1m36.783s
user	0m2.869s
sys	0m1.262s
ubuntu@ubuntu:~/turtlebot4_ws$
```
However, it takes a very long time to execute the action. 
https://github.com/turtlebot/turtlebot4/issues/332 
I found a similar problem, but in my environment, I can't connect the robot to wifi, so I want to speed up action execution with a discovery server.

dgel27, 2024-08-05:
Hi,
How did you solve it? What was a root problem?

RustyCPR, 2024-08-19:
No recent updates. Ticket is closed.

closed 2024-08-19 by RustyCPR

mcvlad, 2024-10-12:
> Hi, How did you solve it? What was a root problem?

@dgel27 just in case this is may be still useful: I encountered the same issue and it turned out to be mismatching `RMW_IMPLEMENTATION` between the host and create3 app config. Once they were set to match (either `rmw_fastrtps_cpp` or `rmw_cyclonedds_cpp`), the mismatching message size error went away, and things started working.