turtlebot/turtlebot4 #455: rviz launch doesnt work

taara67, 2024-08-06:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

turtlebot4_desktop
turtlebot4_viz

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_desktop.html
```
ros2 launch turtlebot4_viz view_robot.launch.py
```
By doing this, the map should appear.

### Actual behaviour

doesnt work

### Error messages

```bash
ubuntu@ubuntu:~$ ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-08-06-09-32-56-248634-ubuntu-22974
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [22975]
connect /tmp/.X11-unix/X10: No such file or directory
[rviz2-1] qt.qpa.xcb: could not connect to display localhost:10.0
[rviz2-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-1]
[rviz2-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-1]
[ERROR] [rviz2-1]: process has died [pid 22975, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_n1sj7ru6 -r /tf:=tf -r /tf_static:=tf_static'].
ubuntu@ubuntu:~$
```
```

### To Reproduce

Installed humble img for rpi4, 
updated the robot firmware to the latest version, 
 installed turtlebot4-desktop, 
 RMW is cyclone 
and logged into the rpi with ssh.

Perhaps the rpi does not have a desktop environment such as xserver.
There is no such procedure in the manual, so is there some other reason?

smatarCPR, 2024-08-09:
Hello @taara67 ,
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry to hear that you are having issues with one of our products.

From the terminal output shared it appears that you are running the RVIZ command whilst SSH'ed to the Turtlebot 4. The Turtlebot 4 desktop metapackage and RVIZ commands should be run on the offboard computer and not on the Raspberry Pi.

You mentioned that you have a discovery server setup, are you able to view and echo topics from the Turtlebot on your remote computer (when not SSH'ed)?

If so, then launching RVIZ should be the same: `ros2 launch turtlebot4_viz view_robot.launch.py` but run on the offboard/remote computer.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

taara67, 2024-08-20:
Thank you for your reply.

I was able to run the view from a remote PC.

https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html

I immediately tried to run SLAM using the above procedure, but
the map does not appear in Rviz.

Is there anything I can do?

Also, although it is not essential, there is an issue where create3 can only connect to the LAN using iPhone tethering.

On my home wifi, a password error is displayed no matter how many times I try.

Because of this issue, I have not tried simple discovery. I would like to solve it.

taara67, 2024-08-20:
I was able to run it successfully after changing the RMW environment variable !
I confirmed the output from the RPlidar and successfully created a map !
thank you

closed 2024-08-26 by RustyCPR