turtlebot/turtlebot4 #456: Rviz2 Generating a Map Issue

tharukakk1, 2024-08-07:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Select One

### Package version

n/a

### Type of issue

LIDAR

### Expected behaviour

SLAM should launch and run without errors. The only messages should be what is shown below. Same error occurs with launching turtlebot4_viz view_robot.launch.py

$ ros2 launch turtlebot4_navigation slam.launch.py
[INFO] [launch]: All log files can be found below /home/turtlebots/.ros/log/2024-08-06-16-27-27-276857-hybridmotion-optiplex-14619
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [14620]
[sync_slam_toolbox_node-1] [INFO] [1722986847.401641807] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1722986847.428373848] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1722986847.428603710] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.

### Actual behaviour

Error messages are continuously returned by SLAM. Message filter dropping line repeated continuously.

$ ros2 launch turtlebot4_navigation slam.launch.py
[INFO] [launch]: All log files can be found below /home/turtlebots/.ros/log/2024-08-06-16-27-27-276857-hybridmotion-optiplex-14619
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [14620]
[sync_slam_toolbox_node-1] [INFO] [1722986847.401641807] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1722986847.428373848] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1722986847.428603710] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
**[sync_slam_toolbox_node-1] [INFO] [1722986860.581604523] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1722986815.054 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1722986860.715839313] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1722986815.188 for reason 'discarding message because the queue is full'**

### Error messages

```bash
[sync_slam_toolbox_node-1] [INFO] [1722986860.581604523] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1722986815.054 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1722986860.715839313] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1722986815.188 for reason 'discarding message because the queue is full'
```

### To Reproduce

1. confirm turtlebot is running and all hardware is connected
2. confirm ros2 topic list is full $ ros2 topic list
3. $ source /opt/ros/humble/setup.bash
4. $ ros2 launch turtlebot4_navigation slam.launch.py

### Other notes

The network does not have internet access due to university restrictions.

smatarCPR, 2024-08-09:
Hello @tharukakk1 ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with one of our products.

In terms of mapping:
1. When you list the topics using ros2 topic list: are the topics on the Turtlebot 4 and your computer match? Are you able to echo any topic and do you get the data? In particular, are you able to echo Lidar data?
2. Would it be possible to launch RVIZ (without mapping) and see if you can visualise the Lidar topic? 
3. Are you launching SLAM on the Turtlebot 4 or on your remote computer? Have you installed `sudo apt install ros-humble-turtlebot4-navigation` on your computer? What is the terminal output when launching navigation slam.launch.py?
4. For the error message you have noted, this could be an indicator that the times on the Turtlebot and your remote computer are not synched. We have the following recommendations with regards to checking timesync:

Setup NTP config Create3 to time sync Create3 and RPi - see instructions in https://github.com/turtlebot/turtlebot4/issues/216.
Setup NTP server on your remote PC on the network - see instructions [here](https://ubuntu.com/server/docs/use-timedatectl-and-timesyncd)
Set up the RPi to sync to that PC
Run the command: systemctl status systemd-timesyncd and see when the timesync last occurred.
For the Raspberry Pi, you can use systemctl status chrony to see when the last sync was and you can force a sync by restarting the chrony service.
Ensure that the Raspberry Pi can ping the Ubuntu NTP servers using ping [ntp.ubuntu.com](http://ntp.ubuntu.com/)

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

tharukakk1, 2024-08-15:
Hi @smatarCPR, sorry for the late reply, I came to the lab on W/R/F this week instead of M/T. I started up the robot and everything worked on the first try. It seems like all of these errors occurred because I was doing other work on the robot last week and didn't shut it down before starting RViz2. I've tried it multiple times now and it works perfectly. It still returns the same time-keeping error but Nav2 and mapping in RViz2 work anyways.

closed 2024-08-15 by tharukakk1