turtlebot/turtlebot4 #458: Impossible to control TB4 with my PC

ValentinLeglise, 2024-08-16:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-desktop
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 1.0.0-1jammy.20240730.174637
Depends: ros-humble-turtlebot4-viz, ros-humble-ros-workspace
Description: Turtlebot4 Desktop Metapackage

### Type of issue

Networking

### Expected behaviour

Follow the setup guide: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html.
Try any network configuration (Simple discovery or discovery Server)
Try to control the robot using a shell on my PC, and not SSH inside the Raspeberry. Trying to use the command: ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
‘linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.0’

### Actual behaviour

I followed the installation guide several times to be able to control my TB4 with my PC via Wifi, not by SSH inside the Rb Pi. Each time, i reflashed the SD card and factory reset the Create3.

I've tried both network configurations (Simple discovery or Discovery Server). I'm currently using Discovery Server and I can control the robot using SSH. However, in a shell on my PC, after sourcing, I can see all the topics after making ros2 topic list:  

vleglise@pc-front:~$ ros2 topic list
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

This proves that i can publish  messages on these topics, but when i try this command, nothing happen: 

ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.0"

The command loop in the shell but the robot doesn't move. I verified all wired connection and remove the battery during few minutes before reinstalling it.

What can i do ?

### To Reproduce

1. Follow guide
2. Choose Discovery Server
3. See all topics in User PC's shell
4. Run ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.0"

smatarCPR, 2024-08-21:
Hello @ValentinLeglise ,
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry to hear that you are having issues with one of our products.

A couple of questions we have regarding the current issues you are facing:

- The cmd_vel command you have sent appears to be missing an extra \ character which typically appears following the geometry_msg/msg/Twist argument (see https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html#command-velocity):
```c
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.0"
``` 

- When viewing the repeated messages after sending a cmd_vel command, do the velocity vectors for angular and linear velocity match the command that has been input? An excerpt from the terminal output when a cmd_vel command is published whilst SSH'ed accompanied with an excerpt from a cmd_vel command from the remote computer would be greatly appreciated.

Looking forward to hearing from you soon.
Best Regards,
Saif M.

aharshac, 2024-09-03:
@ValentinLeglise I have the same problem. @smatarCPR your solution does not work from the PC.

aharshac, 2024-09-03:
I've tried the same motion (once front and then back) on both RPi and PC. The outputs in both cases on both devices with `ros2 topic echo /cmd_vel` are the same!

## Case 1: teleop_twist_keyboard on **rpi**
Echo from rpi:
```
linear:
  x: 0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
linear:
  x: -0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
```

Echo from pc:
```
linear:
  x: 0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
linear:
  x: -0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
```

## Case 2: teleop_twist_keyboard on **PC**
Echo from rpi:
```
linear:
  x: 0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
linear:
  x: -0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
```

Echo from pc:
```
linear:
  x: 0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
linear:
  x: -0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
```

aharshac, 2024-09-03:
@smatarCPR Here's the response of `ros2 topic info -v /cmd_vel` from Rpi and PC in both cases:

## Case 1: teleop_twist_keyboard on **rpi**
Echo from rpi:
```
Type: geometry_msgs/msg/Twist

Publisher count: 2

Node name: teleop_twist_keyboard
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.bf.dd.5e.09.68.35.00.00.00.00.00.00.11.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: teleop_twist_joy_node
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.bf.dd.d0.04.9e.69.00.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: motion_control
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: SUBSCRIPTION
GID: 01.0f.40.4d.fb.05.ec.c4.01.00.00.00.00.00.8d.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

Echo from pc:
```
Type: geometry_msgs/msg/Twist

Publisher count: 2

Node name: teleop_twist_keyboard
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.bf.dd.5e.09.68.35.00.00.00.00.00.00.11.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: teleop_twist_joy_node
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.bf.dd.d0.04.9e.69.00.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: _NODE_NAME_UNKNOWN_
Node namespace: _NODE_NAMESPACE_UNKNOWN_
Topic type: geometry_msgs/msg/Twist
Endpoint type: SUBSCRIPTION
GID: 01.0f.40.4d.fb.05.ec.c4.01.00.00.00.00.00.8d.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

## Case 2: teleop_twist_keyboard on **PC**
Echo from rpi:
```
Type: geometry_msgs/msg/Twist

Publisher count: 2

Node name: teleop_twist_keyboard
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.6f.f6.1f.ee.f2.98.00.00.00.00.00.00.11.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: teleop_twist_joy_node
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.bf.dd.d0.04.9e.69.00.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: motion_control
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: SUBSCRIPTION
GID: 01.0f.40.4d.fb.05.ec.c4.01.00.00.00.00.00.8d.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

Echo from pc:
```
Type: geometry_msgs/msg/Twist

Publisher count: 2

Node name: teleop_twist_keyboard
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.6f.f6.1f.ee.f2.98.00.00.00.00.00.00.11.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: teleop_twist_joy_node
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.bf.dd.d0.04.9e.69.00.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: _NODE_NAME_UNKNOWN_
Node namespace: _NODE_NAMESPACE_UNKNOWN_
Topic type: geometry_msgs/msg/Twist
Endpoint type: SUBSCRIPTION
GID: 01.0f.40.4d.fb.05.ec.c4.01.00.00.00.00.00.8d.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

RustyCPR, 2024-09-13:
Hi @aharshac 
Sorry for the issues you are experiencing. Could you please open a new ticket a new ticket to address your issue. 
If you are trying to drive backwards you set the safety override to backup_only or full in the create3 [webserver](https://iroboteducation.github.io/create3_docs/webserver/application/#:~:text=minutes%20to%20complete.-,Application%20Configuration,-There%20are%20two) to drive backwards.

I'll close this ticket because @ValentinLeglise has not provided a update in some time. 
Thanks,
Rusty

closed 2024-09-13 by RustyCPR

aharshac, 2024-09-19:
@RustyCPR I solved my problem by following https://github.com/turtlebot/turtlebot4/issues/425#issuecomment-2283428277