turtlebot/turtlebot4 #460: User PC ros2topic list issues - discovery mode

swapchal, 2024-08-22:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Windows / MAC

### Built from source or installed?

Installed

### Package version

- 

### Type of issue

Networking

### Expected behaviour

Expected the ros2 topic list to have both create3 and rPi topics when called on the user PC after connecting to discovery server. 

### Actual behaviour

after configuring server on user PC, the ros2 topic list does not have same list as the rPi. 

### Error messages

```bash
~$ wget -qO - https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/turtlebot4_discovery/configure_discovery.sh | bash <(cat) </dev/tty
ROS_DOMAIN_ID [0]: 0
Enter the information for the first discovery server
Discovery Server ID [0]: 0
Discovery Server IP: 192.168.68.77
Discovery Server Port [11811]: 11811
Re-enter the last server (r), add another server (a), or done (d): d
Configuring:
 ROS_DOMAIN_ID=0
 ROS_DISCOVERY_SERVER="192.168.68.77:11811;"
Source your ~/.bashrc file to apply changes
swapn@Swap475:~$ source ~/.bashrc
swapn@Swap475:~$ ros2 daemon stop; ros2 daemon start
The daemon is not running
The daemon has been started
swapn@Swap475:~$ ros2 topic list
/parameter_events
/rosout
swapn@Swap475:~$ ros2 topic list
/parameter_events
/rosout
swapn@Swap475:~$
```

### To Reproduce

1. set up discovery
2. connect user pc to discovery
3. enter ros2 topic list

RustyCPR, 2024-08-26:
Hi @swapchal 
Whats your remote PC set up? are you using a VM on your Windows / Mac PC? 
Can you ping the TB4 from your PC? 
Can you check the ROS domain ID and IP match the TB4? 

Thanks,
Rusty

swapchal, 2024-08-27:
Hi, All sorted Now! But i actually have another issue with Nav2. when launching nav2, i get the  following error:
Message Filter dropping message: frame 'rplidar_link' at time 1724768550.889 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
Im currently using a discovery server, running the remote PC through a VM on my windows. Im not sure how to fix this issue. My create3 also has this message in the logs:
Aug 27 15:05:06 iRobot-67BD9DAA9E89421F9414749033CFA282 user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.004242 is too high, ignoring

swapchal, 2024-08-28:
Update: ive tried to follow the instructions with the ntp server on the create3, and also directly connecting my pc to the Rpi time sync so it is the same. However, the same error messages are still there for the 'rp_lidar' and create3 logs still suggest 'delay is too high, ignoring'. Would appreciate any help on this matter. Thanks

closed 2024-09-02 by swapchal