turtlebot/turtlebot4 #462: Create3 ros2 issues after attempting update

mbergman257, 2024-09-01:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

```
ii  ros-humble-turtlebot4-base                        1.0.3-1jammy.20240731.003557            arm64        Turtlebot4 Base Node
ii  ros-humble-turtlebot4-bringup                     1.0.3-1jammy.20240731.132119            arm64        Turtlebot4 Robot Bringup
ii  ros-humble-turtlebot4-description                 1.0.5-1jammy.20240731.131800            arm64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-diagnostics                 1.0.3-1jammy.20240731.005223            arm64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-msgs                        1.0.5-1jammy.20240730.205330            arm64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                  1.0.5-1jammy.20240731.112247            arm64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                        1.0.5-1jammy.20240731.002306            arm64        Turtlebot4 Node
ii  ros-humble-turtlebot4-robot                       1.0.3-1jammy.20240731.133533            arm64        Turtlebot4 Robot Metapackage
ii  ros-humble-turtlebot4-setup                       1.0.4-1jammy.20240729.103100            arm64        Turtlebot4 setup scripts
ii  ros-humble-turtlebot4-tests                       1.0.3-1jammy.20240730.205422            arm64        Turtlebot4 System Tests
```

### Type of issue

Networking

### Expected behaviour

Create3 topics work, e.g.
`ros2 topic echo /battery_state`
`ros2 topic echo /wheel_vels`
`ros2 topic echo /battery_state`

### Actual behaviour

Something is wrong with my create3 ros2:

- ros2 topic echo gives no response for wheel_vels, battery_state, etc.
- diagnostics window via rqt robot monitor:
[image]

- ros2 topic info wheel_status
[image]

- ros2 node list
[image]

- undock button does nothing

### To Reproduce

1. Be running discovery server on the previous turtlebot4 robot version
2. Update robot via https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#updating-the-turtlebot-4
3. Also tried resetting everything to defaults via create3 webserver factory reset and turtlebot4-setup 'apply defaults', and uninstall/re-install robot upstart

### Other notes

I'm going to try to update on another robot as well.

mbergman257, 2024-09-02:
I have the same results on the second robot. Looks like the create3_republisher node is missing.

mbergman257, 2024-09-06:
I tried starting from scratch:
1. re-flash a new SD card with 1.04 from http://download.ros.org/downloads/turtlebot4/
2. factory reset create3 on latest firmware H.2.6

Now the TB4 is in simple discovery- create3 topics are available, everything seems to work as intended. Now I try to change to discovery server:

3. ssh into TB4
4. do `turtlebot4-setup` with following settings and apply:
[image]
[image]
5. Uninstall, Install, Start TB4 upstart
6. Reboot Create3 & TB4
7. No create3 topics available still
8. I've double checked settings on create3 and pi and they seem like they all applied correctly. Please help!

aharshac, 2024-09-09:
Hi @mbergman257. Maybe this will help https://github.com/turtlebot/turtlebot4/issues/425#issuecomment-2283428277

mbergman257, 2024-09-09:
Thanks for the information, however, I think now my problem is actually two problems. In the discovery server configuration, 1) my create3 topics are not available from the raspberry pi and 2) the create3 republisher isn't starting - it works when I launch it manually. I don't think I have the same problem as mentioned above,  @smatarCPR @RustyCPR can you help please?

mbergman257, 2024-09-12:
I have switched to using simple discovery with unique ros_domain_ids to handle multiple robots running at once, while we wait for a resolution.

mbergman257, 2024-09-16:
https://github.com/turtlebot/turtlebot4/issues/468

closed 2024-09-16 by mbergman257