turtlebot/turtlebot4 #463: Unable to do navigation - Timesync, create3 CPU

swapchal, 2024-09-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Windows / MAC

### Built from source or installed?

Installed

### Package version

unsure

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Able to navigate to goal pose 

### Actual behaviour

After setting the goal pose, I the robot does not move and displays some errors. Also create3 hits 100% usage and is not time synching. (delay too high message)

### Error messages

```bash
1. 
Behavior Tree tick rate 100.00 was exceeded!
Timed out while waiting for action server to acknowledge goal request for compute_path_to_pose
[planner_server-3] [INFO] [1725246766.923489017] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[bt_navigator-5] [WARN] [1725246766.954822490] [BehaviorTreeEngine]: Behavior Tree tick rate 100.00 was exceeded!
[bt_navigator-5] [WARN] [1725246766.954894165] [bt_navigator_navigate_to_pose_rclcpp_node]: Node timed out while executing service call to global_costmap/clear_entirely_global_costmap.
2. Create3 CPU usage is reaching 100% 
3. Create3 timesync is not occuring despite setting the ntpd and following all the steps to sync. it shows up as a client on the Rpi.
```

### To Reproduce

1. Setup TB4
2. Run slam and nav2 
3. set initial pose
4. set goal pose 

### Other notes

I am running on a virtual machine on windows. I have also been trying to set up the turtlebot4 for a few weeks not with quite a few issues so would really appreciate any help to move on with my project. Thanks

smatarCPR, 2024-09-18:
Hello @swapchal 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4.

I have the following recommendations:

- In terms of the Create3 not syncing are you restarting the ntpd on the webserver? Have you modified the ntp.conf on the Create3 per the instructions found here https://github.com/turtlebot/turtlebot4/issues/216 If possible, please try flashing/re-flashing the latest firmware onto the Create3 (https://iroboteducation.github.io/create3_docs/releases/overview/) Should this fail please provide the Create3 logs - https://iroboteducation.github.io/create3_docs/webserver/logs/
- In terms of platform do note that currently the only officially supported setup is to have a machine running Ubuntu natively. Is this option available to you (i.e. a computer with Linux installed as its OS)?

Looking forward to hearing from you soon.
Best Regards,
Saif

swapchal, 2024-10-01:
Was able to sort it out thanks

closed 2024-10-01 by swapchal