turtlebot/turtlebot4 #466: Navigation stack having problem starting up (Smoother server and localization server sometimes giving trouble when trying to launch navigation)

xylhal, 2024-09-05:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Unsure

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I expect the nav2 navigation stack to start up without error so that it could be used for navigation. 

### Actual behaviour

Often times, when I start up the navigation node after running localization, I get the error that the smoother server has failed and will not not respond. 

### Error messages

```bash
[smoother_server.rclcpp]: failed to send response to /smoother_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
```

### To Reproduce

1. Run localization to load map: `ros2 launch turtlebot4_navigation localization.launch.py map:=office.yaml`
2. Run navigation node: `ros2 launch turtlebot4_navigation nav2.launch.py # navigation`

### Other notes

I have tried re-flashing the image, setting up the turtlebot from scratch following the instructions, factory resetting the create3 base and the error still occurs. Most of them time, I have to restart the turtlebot multiple times before it can be loaded successfully. 

I'm currently using the turtlebot4_standard_humble_1.0.0 image, and running it in the discovery server mode. 

I'm also running this with an actual robot and not in simulation.

RustyCPR, 2024-09-13:
Hi @xylhal 
Sorry to see you're encountering issues.
Can you please recreate the issue and provide full terminal output from localization and navigation? 
Thank you,
Rusty

xylhal, 2024-09-13:
```
os2 launch turtlebot4_navigation localization.launch.py map:=maps/demo_map_0822.yaml 
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2024-09-13-13-52-34-263614-iMac-228661
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [228662]
[INFO] [amcl-2]: process started with pid [228664]
[INFO] [lifecycle_manager-3]: process started with pid [228666]
[lifecycle_manager-3] [INFO] [1726260754.518674706] [lifecycle_manager_localization]: Creating
[amcl-2] [INFO] [1726260754.519741823] [amcl]: 
[amcl-2] 	amcl lifecycle node launched. 
[amcl-2] 	Waiting on external lifecycle transitions to activate
[amcl-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-2] [INFO] [1726260754.521076312] [amcl]: Creating
[lifecycle_manager-3] [INFO] [1726260754.521607375] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[map_server-1] [INFO] [1726260754.534376180] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1726260754.534482151] [map_server]: Creating
[lifecycle_manager-3] [INFO] [1726260756.523261554] [lifecycle_manager_localization]: Waiting for service map_server/get_state...
[lifecycle_manager-3] [INFO] [1726260757.548768447] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-3] [INFO] [1726260757.549185823] [lifecycle_manager_localization]: Configuring map_server
[map_server-1] [INFO] [1726260757.551469384] [map_server]: Configuring
[map_server-1] [INFO] [map_io]: Loading yaml file: maps/demo_map_0822.yaml
[map_server-1] [DEBUG] [map_io]: resolution: 0.05
[map_server-1] [DEBUG] [map_io]: origin[0]: -17.7
[map_server-1] [DEBUG] [map_io]: origin[1]: -47.6
[map_server-1] [DEBUG] [map_io]: origin[2]: 0
[map_server-1] [DEBUG] [map_io]: free_thresh: 0.25
[map_server-1] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-1] [DEBUG] [map_io]: mode: trinary
[map_server-1] [DEBUG] [map_io]: negate: 0
[map_server-1] [INFO] [map_io]: Loading image_file: maps/demo_map_0822.pgm
[map_server-1] [DEBUG] [map_io]: Read map maps/demo_map_0822.pgm: 615 X 866 map @ 0.05 m/cell
[lifecycle_manager-3] [INFO] [1726260757.694424180] [lifecycle_manager_localization]: Configuring amcl
[amcl-2] [INFO] [1726260757.694603223] [amcl]: Configuring
[amcl-2] [INFO] [1726260757.694668777] [amcl]: initTransforms
[amcl-2] [INFO] [1726260757.703748645] [amcl]: initPubSub
[amcl-2] [INFO] [1726260757.710451547] [amcl]: Subscribed to map topic.
[lifecycle_manager-3] [INFO] [1726260757.715998960] [lifecycle_manager_localization]: Activating map_server
[map_server-1] [INFO] [1726260757.716179254] [map_server]: Activating
[map_server-1] [INFO] [1726260757.716727144] [map_server]: Creating bond (map_server) to lifecycle manager.
[lifecycle_manager-3] [INFO] [1726260757.921461425] [lifecycle_manager_localization]: Server map_server connected with bond.
[lifecycle_manager-3] [INFO] [1726260757.921690619] [lifecycle_manager_localization]: Activating amcl
[amcl-2] [INFO] [1726260757.923081035] [amcl]: Activating
[amcl-2] [INFO] [1726260757.923336884] [amcl]: Creating bond (amcl) to lifecycle manager.
[amcl-2] [INFO] [1726260758.090311893] [amcl]: Received a 615 X 866 map @ 0.050 m/pix
[amcl-2] [INFO] [1726260758.171956936] [amcl]: createLaserObject
[lifecycle_manager-3] [ERROR] [1726260758.835965461] [lifecycle_manager_localization]: Failed to change state for node: amcl. Exception: transition invoked while in transition.
[lifecycle_manager-3] [ERROR] [1726260758.836363692] [lifecycle_manager_localization]: Failed to bring up all requested nodes. Aborting bringup.
```
This above is the output from running `ros2 launch turtlebot4_navigation nav2.launch.py`

The localization usually runs fine without throwing any errors. 
```
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2024-09-13-13-52-34-263614-iMac-228661
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [228662]
[INFO] [amcl-2]: process started with pid [228664]
[INFO] [lifecycle_manager-3]: process started with pid [228666]
[lifecycle_manager-3] [INFO] [1726260754.518674706] [lifecycle_manager_localization]: Creating
[amcl-2] [INFO] [1726260754.519741823] [amcl]: 
[amcl-2] 	amcl lifecycle node launched. 
[amcl-2] 	Waiting on external lifecycle transitions to activate
[amcl-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-2] [INFO] [1726260754.521076312] [amcl]: Creating
[lifecycle_manager-3] [INFO] [1726260754.521607375] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[map_server-1] [INFO] [1726260754.534376180] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[... 75 lines cut ...]
[amcl-2] [INFO] [1726260841.540248136] [amcl]: Setting pose (1726260841.540247): 0.451 -24.106 -3.071
[amcl-2] [INFO] [1726260849.029340696] [amcl]: initialPoseReceived
[amcl-2] [WARN] [1726260849.029695591] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1726260849.029562 but the latest data is at time 1726260848.980560, when lo [... cut ...]
[amcl-2] [INFO] [1726260978.373043295] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1726260977.055 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1726260978.528863363] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1726260977.436 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1726260978.718702148] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1726260977.558 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
```

xylhal, 2024-10-09:
Hi @RustyCPR, @smatarCPR 

Any feedback on this? I realized that this is a common issue but this issue always gets closed. I would like to dive deeper into the cause of this. 

Thanks

smatarCPR, 2024-10-11:
Hello @xylhal
Apologies for the delay on this we are still processing the data from the logs, but we will provide an update soon.
Best Regards,
Saif

daisukes, 2024-10-15:
Hi @xylhal, @RustyCPR, and @smatarCPR 

I recently got into the same issue with my robot.

> [lifecycle_manager-3] [ERROR] [1726260758.835965461] [lifecycle_manager_localization]: Failed to change state for node: amcl. Exception: transition invoked while in transition.

I found that this error is raised at this line, and it may be a timing-related issue while creating a bond between a lifecycle manager and a node, in this case, amcl.

https://github.com/ros/bond_core/blob/52e181c717186185cb25fed62ed95a16804b5a17/smclib/include/smclib/statemap.hpp#L168

I plan to look into it, but I just wanted to tell you what I found.
Best regards,

daisukes, 2024-10-15:
https://github.com/ros/bond_core/pull/97

I made a PR to `bond_core`, which may fix this issue.
Best,

xylhal, 2024-10-17:
@daisukes Thank you, please let us know if you have any update on whether your PR fixed the issue.

@RustyCPR, @smatarCPR Please also let me know what your findings are

hilary-luo, 2024-11-01:
@xylhal the logs that you provided are all from the localization launch, can you please share the log from the nav2 launch command that you are using? Additionally, can you clarify where you are running the localization / nav nodes? We recommend running them offboard, on a separate machine that runs Ubuntu natively (not on the turtlebot itself). 

@daisukes I took a quick look at your PR and although your comment says that you want the change included on Humble, the PR is for Galactic. Galactic is end of life and is no longer supported, if you want them to look at your PR I would suggest creating a new one for Humble.

daisukes, 2024-11-01:
> @daisukes I took a quick look at your PR and although your comment says that you want the change included on Humble, the PR is for Galactic. Galactic is end of life and is no longer supported, if you want them to look at your PR I would suggest creating a new one for Humble.

Thanks for the check!
There is no humble branch in bond_core, and humble use 3.0.2, equivalent to the galactic branch.
I've tried the ros2 branch, but it might have an API change and won't work with humble nav2.
Do you have any suggestions?

https://github.com/ros2-gbp/bond_core-release/tree/master?tab=readme-ov-file#bond_core-humble---302-3

xylhal, 2024-11-01:
@hilary-luo Apologies, here is the screenshot of the error we faced when running nav2 launch command. Although the error above from running the localization sometimes occurs as well, but most of the them, the error comes from launching nav2. And I am currently running them offboard on a dedicated Ubuntu machine following the discovery server architecture
```
~/turtlebot4_ws$ ros2 launch turtlebot4_navigation nav2.launch.py
[INFO] [launch]: All log files can be found below /home/.ros/log/2024-11-01-12-53-01-563861-iMac-797603
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [797604]
[INFO] [smoother_server-2]: process started with pid [797606]
[INFO] [planner_server-3]: process started with pid [797608]
[INFO] [behavior_server-4]: process started with pid [797610]
[INFO] [bt_navigator-5]: process started with pid [797612]
[INFO] [waypoint_follower-6]: process started with pid [797614]
[INFO] [velocity_smoother-7]: process started with pid [797616]
[INFO] [lifecycle_manager-8]: process started with pid [797618]
[planner_server-3] [INFO] [1730490781.976843993] [planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-2] [INFO] [1730490781.979593777] [smoother_server]: 
[smoother_server-2] 	smoother_server lifecycle node launched. 
[smoother_server-2] 	Waiting on external lifecycle transitions to activate
[smoother_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1730490781.993174885] [bt_navigator]: 
[bt_navigator-5] 	bt_navigator lifecycle node launched. 
[bt_navigator-5] 	Waiting on external lifecycle transitions to activate
[bt_navigator-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1730490781.993324240] [bt_navigator]: Creating
[smoother_server-2] [INFO] [1730490781.998475878] [smoother_server]: Creating smoother server
[planner_server-3] [INFO] [1730490782.000698533] [planner_server]: Creating
[waypoint_follower-6] [INFO] [1730490782.002810615] [waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[velocity_smoother-7] [INFO] [1730490782.007289702] [velocity_smoother]: 
[velocity_smoother-7] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-7] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1730490782.010565170] [waypoint_follower]: Creating
[controller_server-1] [INFO] [1730490782.012620460] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1730490782.015495857] [global_costmap.global_costmap]: 
[planner_server-3] 	global_costmap lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[behavior_server-4] [INFO] [1730490782.016740095] [behavior_server]: 
[behavior_server-4] 	behavior_server lifecycle node launched. 
[behavior_server-4] 	Waiting on external lifecycle transitions to activate
[behavior_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1730490782.021188771] [controller_server]: Creating controller server
[planner_server-3] [INFO] [1730490782.022262209] [global_costmap.global_costmap]: Creating Costmap
[lifecycle_manager-8] [INFO] [1730490782.022583267] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-8] [INFO] [1730490782.024381683] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[controller_server-1] [INFO] [1730490782.027705975] [local_costmap.local_costmap]: 
[controller_server-1] 	local_costmap lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1730490782.028262660] [local_costmap.local_costmap]: Creating Costmap
[lifecycle_manager-8] [INFO] [1730490784.025644963] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1730490786.026172570] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1730490786.149798928] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-8] [INFO] [1730490786.149847865] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-1] [INFO] [1730490786.150178719] [controller_server]: Configuring controller interface
[controller_server-1] [INFO] [1730490786.150822070] [controller_server]: getting goal checker plugins..
[... cut ...]
[smoother_server-2] [INFO] [1730490791.474019006] [smoother_server]: Smoother Server has simple_smoother  smoothers available.
[smoother_server-2] [WARN] [1730490791.588452217] [smoother_server.rclcpp]: failed to send response to /smoother_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
[controller_server-1] [INFO] [1730490798.341902004] [local_costmap.local_costmap]: StaticLayer: Resizing static layer to 615 X 866 at 0.050000 m/pix
```

xylhal, 2024-11-01:
@hilary-luo This is from a different session of re-running the nav2 launch file
```
`~/turtlebot4_ws$ ros2 launch turtlebot4_navigation nav2.launch.py
[INFO] [launch]: All log files can be found below /home/.ros/log/2024-11-01-13-03-28-962151-iMac-798691
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [798692]
[INFO] [smoother_server-2]: process started with pid [798694]
[INFO] [planner_server-3]: process started with pid [798696]
[INFO] [behavior_server-4]: process started with pid [798698]
[INFO] [bt_navigator-5]: process started with pid [798700]
[INFO] [waypoint_follower-6]: process started with pid [798702]
[INFO] [velocity_smoother-7]: process started with pid [798704]
[INFO] [lifecycle_manager-8]: process started with pid [798706]
[behavior_server-4] [INFO] [1730491409.382424999] [behavior_server]: 
[behavior_server-4] 	behavior_server lifecycle node launched. 
[behavior_server-4] 	Waiting on external lifecycle transitions to activate
[behavior_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1730491409.387354568] [planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[velocity_smoother-7] [INFO] [1730491409.389075560] [velocity_smoother]: 
[velocity_smoother-7] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-7] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1730491409.390346802] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1730491409.394796486] [planner_server]: Creating
[controller_server-1] [INFO] [1730491409.396629725] [controller_server]: Creating controller server
[smoother_server-2] [INFO] [1730491409.398142869] [smoother_server]: 
[smoother_server-2] 	smoother_server lifecycle node launched. 
[smoother_server-2] 	Waiting on external lifecycle transitions to activate
[smoother_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-2] [INFO] [1730491409.402195659] [smoother_server]: Creating smoother server
[controller_server-1] [INFO] [1730491409.404122805] [local_costmap.local_costmap]: 
[controller_server-1] 	local_costmap lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1730491409.405062683] [local_costmap.local_costmap]: Creating Costmap
[planner_server-3] [INFO] [1730491409.409072905] [global_costmap.global_costmap]: 
[planner_server-3] 	global_costmap lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1730491409.409607709] [global_costmap.global_costmap]: Creating Costmap
[lifecycle_manager-8] [INFO] [1730491409.415954329] [lifecycle_manager_navigation]: Creating
[waypoint_follower-6] [INFO] [1730491409.416887030] [waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1730491409.417477151] [waypoint_follower]: Creating
[lifecycle_manager-8] [INFO] [1730491409.418233154] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[bt_navigator-5] [INFO] [1730491409.419928671] [bt_navigator]: 
[bt_navigator-5] 	bt_navigator lifecycle node launched. 
[bt_navigator-5] 	Waiting on external lifecycle transitions to activate
[bt_navigator-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1730491409.420038891] [bt_navigator]: Creating
[lifecycle_manager-8] [INFO] [1730491411.419620761] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1730491413.431492682] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-8] [INFO] [1730491413.431895578] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-1] [INFO] [1730491413.435375529] [controller_server]: Configuring controller interface
[controller_server-1] [INFO] [1730491413.439997063] [controller_server]: getting goal checker plugins..
[controller_server-1] [INFO] [1730491413.451373748] [controller_server]: Controller frequency set to 20.0000Hz
[controller_server-1] [INFO] [1730491413.455163536] [local_costmap.local_costmap]: Configuring
[controller_server-1] [INFO] [1730491413.494030208] [local_costmap.local_costmap]: Using plugin "static_layer"
[... cut ...]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[velocity_smoother-7] [INFO] [1730491424.770654215] [rclcpp]: signal_handler(signum=2)
[velocity_smoother-7] [INFO] [1730491424.771421494] [velocity_smoother]: Running Nav2 LifecycleNode rcl preshutdown (velocity_smoother)
[lifecycle_manager-8] [INFO] [1730491424.770654233] [rclcpp]: signal_handler(signum=2)
[waypoint_follower-6] [INFO] [1730491424.770803187] [rclcpp]: signal_handler(signum=2)
[planner_server-3] [INFO] [1730491424.771104690] [rclcpp]: signal_handler(signum=2)
`
```

xylhal, 2024-11-13:
Hi @RustyCPR @smatarCPR @hilary-luo Do you have updates on your findings? We got a second turtlebot that is setup in a totally independent network and this is still happening.

daisukes, 2024-11-13:
Hi @xylhal 

Could you try to clone my [bond_cpp fix](https://github.com/CMU-cabot/bond_core/tree/galactic-fix) into your workspace and build it with your packages to see if it can fix the issue?

If it works, please leave a comment for support at the PR!

- https://github.com/ros/bond_core/pull/97

Thanks

xylhal, 2024-11-13:
@daisukes Sure, I will try it out and get back to you

hilary-luo, 2024-11-25:
@daisukes you could try bringing it up on ROS discourse or discord to try and get some attention to your PR. 

@xylhal was @daisukes solution able to address the issue? If not I would suggest bringing this forward to the Nav2 team to see if they can provide additional insight.

xylhal, 2024-11-25:
@hilary-luo I haven't had time to try @daisukes solution yet, I will update if it works or not. In the meantime, did your team find anything?

RustyCPR, 2025-01-27:
Hi @xylhal  Any progress with this issue? 
Thanks,
Rusty

daisukes, 2025-01-30:
Hi @RustyCPR 

The ROS2 team has decided to use the latest bond_core for release.
It should be distributed soon, and the issue will be fixed.

- https://github.com/ros/bond_core/pull/97#issuecomment-2612486703

To try the fix now, you must build both navigation2 and bond_core from the source.
navigation2 release binary + bond_core@ros2 branch source build won't work due to ABI change.

xylhal, 2025-01-31:
hi @RustyCPR I ended up not trying @daisukes solution yet as we had to proceed with other higher priority experiments with the robot, but the issue is still occurs pretty frequently

@daisukes That's great news! Thanks for the update

@RustyCPR also when the latest version is release, do you guys plan to incorporate those changes into the Turtlebot's packages?

hilary-luo, 2025-02-01:
@xylhal Turtlebot4 uses those packages directly (as dependencies) so once they are available on the build farm you just need to update your packages. No other action is required directly for Turtlebot4 to use these types of changes.

beardyFace, 2025-02-17:
Which versions of the nav2 and core-bond will resolve the timeout issue (if they have been released)? Or do we still need to install both by source? I am using 

ros-humble-navigation2/jammy,now 1.1.18-1jammy.20250210.230037 amd64 [installed,automatic]

ros-humble-bond-core/jammy 4.1.2-1jammy.20250124.133242 amd64

and still getting the same issue with the smoother server timeout

daisukes, 2025-02-18:
This issue is not about timeout but a deadlock.

Have you tried to increase `bond_timeout`? I use `bond_timeout=15.0` to increase timeout from [the default timeout (4.0)](https://github.com/ros-navigation/navigation2/blob/4694386872365851c840737515f89e3d409c1d9f/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L44).

beardyFace, 2025-02-18:
Is the only way to increase the bond_timeout via installing the Nav2 package by source? or is there a configuration that can be passed down via the Turtlebot4 launch files?

beardyFace, 2025-02-20:
@daisukes I installed Nav2 from source (https://docs.nav2.org/development_guides/build_docs/index.html) and adjusted the bond_timeout parameter to 15, 100, 200 and still have run into the same issue. 

I have confirmed via ros2 param get that the bond_timeout is being set correctly, but it is still leading to the "failed to send response to ..." most often at the smoother_server

daisukes, 2025-02-20:
@beardyFace  

- https://github.com/ros-navigation/navigation2/issues/4552#issuecomment-2629255915

It may be related to this.
They think this is a problem specific to turtle bots, but we need to convince them it is a general issue that can happen with low-resource robots.