turtlebot/turtlebot4 #467: Using namespaces adds "_do_not_use" implicitly to create 3 nodes

nisarganc, 2024-09-06:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Create 3 firware: H.2.6
Raspberry PI packages:  Version: 1.0.5-1jammy.20240731.002306

### Type of issue

Other

### Expected behaviour

I am working with three TurtleBot4 Standard model robots. I configured all of them using the Discovery server with unique namespaces and onboard server IDs for each. After applying the settings using the turtlebot4-setup command, create 3 web server configurations were automatically set with buggy namespace: "/turtle2/_do_not_use". As a result, when updating ROS parameters, I have to use  "/turtle2/_do_not_use/motion_control" for node name.

### Actual behaviour

Create 3 web server ros namespace configuration should be automatically set with the same namespace as provided in RPi.

### To Reproduce

1. Update create 3 to latest firmware version
2. Update all packages on Raspberry Pi
3. Configure 2 or more turtlebot4 with DIscovery Server
4. Check the Create 3 Web server for ros 2 namespace

mbergman257, 2024-09-06:
Not a solution to your problem, but here is some information regarding the _do_not_use namespace: 
https://turtlebot.github.io/turtlebot4-user-manual/software/create3.html#create-3-republisher
it is a part of the update adding the create3 republisher, used in the discovery server configuration

nisarganc, 2024-09-10:
@mbergman257 Thanks fa lot for this link! This explains the behavior and helps me in further optimize the topics being republished from RPi :)

hilary-luo, 2024-09-13:
@nisarganc  In case there remains any confusion, please do not use any of the topics that are indicated as "_do_not_use", they are hidden and named as such for a reason. Instead please use the republished copies of these topics. That link has the full description but in short for anyone who comes across this issue, the republished topics are much more robust and allow for running large fleets of turtlebots without overloading the create3 CPU.

closed 2024-09-13 by hilary-luo

nisarganc, 2024-09-16:
@hilary-luo Please look at the screen attached. As you suggested, I used the re-published topic and removed "_do_not_use" from the topic name in the app config of the Create3 web server. However, the reflexes are still active despite setting the ROS parameter to false, and the bumper isn't deactivating.
[image]

hilary-luo, 2024-09-16:
@nisarganc Thanks for bringing this up. If you are setting parameters in that box on the create3 webserver then you likely need to use the "turtle6/_do_not_use" namespace. Everywhere else, offboard the create3 though should be using the republished topics for all interactions. Can you confirm that that works for you and I will make sure the documentation gets updated with a note regarding that box.

nisarganc, 2024-09-18:
@hilary-luo Thank you for the clarification. Yes, parameters setting works with the namespace: "turtle6/_do_not_use".