turtlebot/turtlebot4 #468: Create3 republisher node not running

mbergman257, 2024-09-06:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ii  ros-humble-turtlebot4-base                        1.0.3-1jammy.20240731.003557            arm64        Turtlebot4 Base Node
ii  ros-humble-turtlebot4-bringup                     1.0.3-1jammy.20240731.132119            arm64        Turtlebot4 Robot Bringup
ii  ros-humble-turtlebot4-description                 1.0.5-1jammy.20240731.131800            arm64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-diagnostics                 1.0.3-1jammy.20240731.005223            arm64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-msgs                        1.0.5-1jammy.20240730.205330            arm64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                  1.0.5-1jammy.20240731.112247            arm64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                        1.0.5-1jammy.20240731.002306            arm64        Turtlebot4 Node
ii  ros-humble-turtlebot4-robot                       1.0.3-1jammy.20240731.133533            arm64        Turtlebot4 Robot Metapackage
ii  ros-humble-turtlebot4-setup                       1.0.4-1jammy.20240729.103100            arm64        Turtlebot4 setup scripts
ii  ros-humble-turtlebot4-tests                       1.0.3-1jammy.20240730.205422            arm64        Turtlebot4 System Tests

### Expected behaviour

When using discovery server, create3 topics are published and available. e.g., 
`ros2 topic echo /battery_state`
`ros2 topic echo /wheel_vels`
`ros2 topic echo /wheel_status`

### Actual behaviour

1. Create3 topics are not available via ros2 topic echo
2. ros2 topic info shows no publisher for create3 topic
3. no create3 republisher node is found in ros2 node list
4. undock button does not work

See https://github.com/turtlebot/turtlebot4/issues/462 for screenshots of behavior

### To Reproduce

1. Update to latest turtlebot4 packages, or flash pi with latest packages.
2. Use turtlebot4-setup to configure discovery server according to https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html

mbergman257, 2024-09-16:
I was able to make the discovery server configuration work by giving the robot a namespace through the turtlebot4-setup command. May be related to https://github.com/turtlebot/turtlebot4/issues/425#issuecomment-2283428277, but I'm unsure if the issue is the same, or if they are just similar issues. Unfortunately, I don't have time to look into it more. For the time being, I will be using namespaces.

hilary-luo, 2024-09-16:
@mbergman257 Currently there is a bug with Discovery Server that requires a namespace to be present. The fix has already been merged in but is pending iRobot's next release of the create3_republisher.

Simba185, 2025-02-04:
@hilary-luo any updates on this?

hilary-luo, 2025-02-04:
@Simba185 If I remember correctly the fix on Humble is still only available from source but the fix was released in Jazzy to debian.

closed 2025-02-04 by hilary-luo

Jannkar, 2025-02-11:
Hi! Thanks for providing the namespace workaround. I was also struggling with this for quite a while, and this fixed the issue.

When will this fix be released for Humble as a Debian package? To me, this seems like a huge issue if users who use the discovery server with the default settings (no namespace) can't get it working correctly by following the user manual.