turtlebot/turtlebot4 #469: Can't control the robot while connected to a PC

ValentinLeglise, 2024-09-09:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

*

### Type of issue

Networking

### Expected behaviour

My client follows the setup guide: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html.
Selected Discovery Server configuration
Tried to control the robot using a shell on my PC, and not SSH inside the Raspeberry. Trying to move the  robot with "ros2 run teleop_twist_keyboard teleop_twist_keyboard" o " ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
‘linear:
x: 0.0
y: 0.0
z: 0.0
angular:
x: 0.0
y: 0.0
z: 1.0’

The robot should move.

### Actual behaviour

My customer followed the installation guide several times to be able to control my TB4 with my PC via Wifi, not by SSH inside the Rb Pi. Each time, he reflashed the SD card and factory reset the Create3.

He has tried both network configurations (Simple discovery or Discovery Server). He is currently using Discovery Server and he can't control the robot using SSH or just with his PC. However, in a shell on my PC, after sourcing, he  can see all the topics after making ros2 topic list:

vleglise@pc-front:~$ ros2 topic list
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

He also can see the IP adress with "ros2 topic echo /ip"

This proves that he can publish messages on these topics, but when he tries this command, nothing happen:

ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
x: 0.0
y: 0.0
z: 0.0
angular:
x: 0.0
y: 0.0
z: 1.0"

The command loop in the shell but the robot doesn't move. He verified all wired connection and remove the battery during few minutes before reinstalling it.

He did a diagnosis by following this: https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/diagnostics.html

and the result is: 
[image]

This requires it to put the robot on the recharging station, but it's already there and there are still critical messages about the battery (even though it's a new robot).

If he tries (even directly on the robot) to type : 
ros2 topic echo /battery_state

it gets stuck and nothing is displayed.

What can he does ? 

### To Reproduce

    Follow guide
    Choose Discovery Server
    See all topics in User PC's shell
    Run ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
    "linear:
    x: 0.0
    y: 0.0
    z: 0.0
    angular:
    x: 0.0
    y: 0.0
    z: 1.0"

ValentinLeglise, 2024-09-12:
Hello @RustyCPR, @smatarCPR.

Would it be possible to have an answer. This problem is quite urgent and we would like to solve it asap. Thank you.

RustyCPR, 2024-09-13:
Hi @ValentinLeglise 
Sorry for the delay in replying. 

1. Please ask the customer to verify the [diagnostics](https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/diagnostics.html#diagnostics) is turned on.  If the customer is making mistakes while running the Set up Tool the issue(s) should be identified. 
2. Verify if the customer can see  `ros2 topic echo /battery_state `
3. Also please verify the customer can see `ros2 topic echo /cmd_vel` on the robot when publishing cmd_vel. 
Thank you,
Rusty

mbergman257, 2024-09-16:
@ValentinLeglise A workaround that resolved my issue was to give the robot a namespace with the turtlebot4-setup command. I believe we had the same or a similar issue. Maybe it will help.

ValentinLeglise, 2024-09-17:
Hello @mbergman257 ! What namespace did you used ? Does it have to respect a specific name ?

Thank you for your help.

mbergman257, 2024-09-17:
@ValentinLeglise It didn't need a specific name, it just needed a namespace due to a bug in the create3_republisher used for discovery server. https://github.com/turtlebot/turtlebot4/issues/468

ValentinLeglise, 2024-09-18:
@mbergman257 I will suggest it to my customer. I keep you informed. Thank you

ValentinLeglise, 2024-09-19:
Hello,

@mbergman257 thank you for your help. It works for the moment. My customer can control the robot.

But he informed me that it would be wise to modify the documentation accordingly, as the namespace has been changed and the commands need to be adapted. 
For example, "_ros2 run teleop_twist_keyboard teleop_twist_keyboard **--ros-args -r __ns:=/turtlebot_icam**_"

So if this problem is to remain, it would be wise to specify in the documentation that namespace is to be used, and to specify command syntax taking namespace into account. @RustyCPR.

Thank you both of you.

closed 2024-10-30 by RustyCPR