turtlebot/turtlebot4 #470: How to see the smooth camera image from OKA-D camera?

codingBigBird, 2024-09-09:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

git clone https://github.com/turtlebot/turtlebot4.git -b humble

### Type of issue

Camera

### Expected behaviour

The image from the camera seen in RVIZ is very smooth.
I can see image from the topic: /oakd/rgb/preview/image_raw/compressedDepth
How can I improve this?

### Actual behaviour

There is 10 seconds delay from the camera when the TB4 runs nothing. And when the robot is navigating, I can hardly see the image from camera. Is this because of the limitation of Raspberry pi? 
And Why I can't see any message from /oakd/rgb/preview/image_raw/compressedDepth this topic?
[image]

### To Reproduce

ros2 topic echo /oakd/rgb/preview/image_raw/compressedDepth
(get nothing from the topic)
ros2 launch turtlebot4_viz view_robot.launch.py
(have a look at the camer topic: /oakd/rgb/preview/image_raw)

smatarCPR, 2024-09-18:
Hello @codingBigBird 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot.

In terms of the issue described, it appears that topic you have selected (compressedDepth) is a depth image topic, but there appears to be no RGBD topics present. This may explain why the image feed is very choppy as you reported. To mitigate this you can either:

_Enable RGBD to generate Depth Images_ - This comment (https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838) has some details on how to enable stereo for the Oak-D camera. You can also refer to the depthAI examples for additional node (https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#oak-d https://github.com/luxonis/depthai-ros)

_Use the standard RGB Image_ - Rather than echoing  /oakd/rgb/preview/image_raw/compressedDepth echo  /oakd/rgb/preview/image_raw/compressed

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

codingBigBird, 2024-09-21:
Hi @smatarCPR,
Thanks for replying me. There are eight to ten seconds delay to see the video stream in RVIZ at topic /Robot1/oakd/rgb/preview/image_raw. Is it any method to improve it to one second? I try simple discovery and discovery server. Both of them have this issue.

codingBigBird, 2024-09-24:
And why sometimes my laptop has the topics of the TB4, but I can't see any content of the topic by using ros2 topic echo /topic in Discovery Server.

smatarCPR, 2024-09-27:
Hello,
I have the following recommendations with regards to your queries:

- For the slow feed, when you use the topic /Robot1/oakd/rgb/preview/image_raw this is the uncompressed version of the camera stream. This stream requires a lot of network bandwidth to stream without delay. It is for this reason that if you wish for a stream with minimal delay to use ros2 topic echo /oakd/rgb/preview/image_raw/compressed

With regards to the topics being available but no data being published there could be a number of reasons:
- The topics have not published any data yet
- If you wait for 2 minutes, does that data from the topics become available? If so, then in this case this would imply that the network you are on has a great deal of network traffic.
- There may be a network setting that is blocking traffic between your computer and the Turtlebot. If you connect to the Turtlebot using an ethernet connection (ensuring to update the discovery server settings with the new networking information) are you able to echo topic data?

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-10-30:
Hello @codingBigBird ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-10-30 by smatarCPR