turtlebot/turtlebot4 #472: Clear instructions for Offline configuration and opperation

Simba185, 2024-09-12:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

not applicable,

### Type of issue

Networking

### Expected behaviour

Connect to wired connection for packages and then connect to Wi-Fi for operation.

### Actual behaviour

We need a clear method to utilize the Turtlebot 4 without internet connection.

Is it possible to configure the tb4 with a wired connection then connected to a local wireless network for operation.

If so please provide clear instructions .

thanks.

### To Reproduce

not currently applicable.

mbergman257, 2024-09-12:
Hi, these might be helpful to access the raspberry pi without connecting it to a network:
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#connect-to-the-access-point
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi

RustyCPR, 2024-10-30:
Hi @Simba185 
Did @mbergman257 suggestion help with interfacing to the RPi using ethernet? 

Thanks @mbergman257 ! 

Regards,
Rusty

Simba185, 2024-12-03:
@RustyCPR 

No unfortunately I am really looking for a system that would allow us to configure and use the turtlebot on a wifi network without internet.

 I would need a way to host any files that are needed from the internet, and some method to tell the turtlebot that's where to get the files from.

This is still not an ideal setup. and is not an issue with the turtlebot, but an issues of access within my organization.

Any ideas are appreciated, Thanks.

RustyCPR, 2025-01-20:
Hi Simba, 
Reviewing with our team we have no concerns with running your TB4 on a Wi-Fi network with no internet access. 
Only time the TB4 would need to connect to the internet would be to download packages you require. 

To run on WiFi network with no internet modify chrony.conf on the RPi to point at your own server. https://ubuntu.com/server/docs/how-to-serve-the-network-time-protocol-with-chrony

We don't recommend setting up a local ROS package mirror - connect to the internet as needed to pull packages. 
Best regards,
Rusty

RustyCPR, 2025-08-13:
Hi @Simba185 
I hope this email finds you well.
I will go ahead and close this ticket. Should you need further assistance, please open a new ticket.
Best regards,
Rusty