turtlebot/turtlebot4 #473: Missing Subscriber, Connection between RPi4 and Create3 via USB

TB4User, 2024-09-19:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

PC:
dpkg -l | grep turtlebot4
ii  ros-humble-turtlebot4-description                  1.0.5-1jammy.20240822.152955                      amd64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-desktop                      1.0.0-1jammy.20240822.174439                      amd64        Turtlebot4 Desktop Metapackage
ii  ros-humble-turtlebot4-ignition-bringup             1.0.2-1jammy.20240824.193140                      amd64        TurtleBot 4 Ignition Simulator bringup
ii  ros-humble-turtlebot4-ignition-gui-plugins         1.0.2-1jammy.20240728.203144                      amd64        Turtlebot4 Ignition Simulator GUI Plugins
ii  ros-humble-turtlebot4-ignition-toolbox             1.0.2-1jammy.20240728.225150                      amd64        Turtlebot4 Ignition Toolbox
ii  ros-humble-turtlebot4-msgs                         1.0.5-1jammy.20240728.215722                      amd64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                   1.0.5-1jammy.20240824.192601                      amd64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                         1.0.5-1jammy.20240728.225044                      amd64        Turtlebot4 Node
ii  ros-humble-turtlebot4-simulator                    1.0.2-1jammy.20240824.193207                      amd64        TODO: Package description
ii  ros-humble-turtlebot4-viz                          1.0.0-1jammy.20240822.174054                      amd64        Visualization launchers and helpers for Turtlebot4

RPi-4:
dpkg -l | grep turtlebot4
ii  ros-humble-turtlebot4-base                        1.0.3-1jammy.20240731.003557            arm64        Turtlebot4 Base Node
ii  ros-humble-turtlebot4-bringup                     1.0.3-1jammy.20240824.181335            arm64        Turtlebot4 Robot Bringup
ii  ros-humble-turtlebot4-description                 1.0.5-1jammy.20240822.152608            arm64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-diagnostics                 1.0.3-1jammy.20240731.005223            arm64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-msgs                        1.0.5-1jammy.20240730.205330            arm64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                  1.0.5-1jammy.20240824.203516            arm64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                        1.0.5-1jammy.20240731.002306            arm64        Turtlebot4 Node
ii  ros-humble-turtlebot4-robot                       1.0.3-1jammy.20240824.182049            arm64        Turtlebot4 Robot Metapackage
ii  ros-humble-turtlebot4-setup                       1.0.4-1jammy.20240729.103100            arm64        Turtlebot4 setup scripts
ii  ros-humble-turtlebot4-tests                       1.0.3-1jammy.20240730.205422            arm64        Turtlebot4 System Tests

RPi4 Image flashed to: 
turtlebot4_lite_humble_1.0.4 (latest)

Create3-Version:
H.2.6 (Humble latest)

### Type of issue

Networking

### Expected behaviour

The TB4 is not Executing any Nodes. (Missing Subscriber)
The TB4 is not Executing any Actions (Waiting for an action server to become available...)
After configuring a Republisher from following:
https://github.com/iRobotEducation/create3_examples/tree/humble/create3_republisher

Even if I try to run the Nodes/Actions on the RPi4 itself in my ssh-session.

Used nodes:
ros2 run turtlebot4_tests ros_tests
ros2 run teleop_twist_keyboard teleop_twist_keyboard
[... 164 lines cut ...]
```

### To Reproduce

1. ssh ubuntu@ip
2. ros2 run turtlebot4_tests ros_tests

### Other notes

The discovery server is setup like the description (TB4 Basic Setup)
and should generally run on the RPi4 correctly. 
Running talker/listener testnodes between PC and RPi4 worked well. 

It looks like the RPi4 and the Create3 are not communicating correctly via usb because of the missing Subscriber.
The Button on the Create3 itself is set to Data per USB (no bluetooth). 

Robot Name is blank. 
The Create3 got automatically following namespace: 
/_do_not_use

The Clocks are not synchronised. 
The all red lines of the Create3Logs are shown in the picture, the Logs are copied as .txt

[image]
[Create3Log.txt](https://github.com/user-attachments/files/17057215/Create3Log.txt)

websquadpro, 2024-09-19:
maybe this will help
https://bit.ly/4evXrQ6

you may need to install the c compiler

smatarCPR, 2024-09-27:
Hello @TB4User 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are encountering issues with your Turtlebot 4 unit. In terms of specific advice I can offer with regards to this issue:

- It is recommended to add a namespace to your ROS environment., especially when using a discovery server setup. This will most likely address all of your currently ongoing issues.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saiif

smatarCPR, 2024-10-30:
Hello @TB4User ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-10-30 by smatarCPR