turtlebot/turtlebot4 #476: Lidar hardware issues

alexswerner, 2024-09-23:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

not relevant here

### Expected behaviour

Correct output on `/scan`

### Actual behaviour

Robot "uwbot-02" has the issue that the lidar does start spinning once I subscribe to the /scan topic, but the Lidar only reports Inf. Also, the laser light is not visible.

Robot "uwbot-03" the driver reports an error during startup and does not publish the topic.

### Error messages

```bash
This is the output of the driver log for uwbot-02
Sep 19 20:29:11 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726777751.917515074] [rplidar_composition]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version: '1.12.0'
Sep 19 20:29:14 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726777754.497142925] [rplidar_composition]: RPLIDAR S/N: E6BDED93C0EA98C9A5E698F273634669
Sep 19 20:29:14 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726777754.497312090] [rplidar_composition]: Firmware Ver: 1.29
Sep 19 20:29:14 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726777754.497357034] [rplidar_composition]: Hardware Rev: 7
Sep 19 20:29:14 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726777754.499335465] [rplidar_composition]: RPLidar health status : '0'
Sep 19 20:36:39 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726778199.758821573] [rplidar_composition]: Start
Sep 19 20:36:40 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726778200.303367897] [rplidar_composition]: current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_comp
ensate: 2, flip_x_axis 0
Sep 19 20:36:59 uwbot-02 turtlebot4-start[2472]: [rplidar_composition-5] [INFO] [1726778219.398541777] [rplidar_composition]: Stop

On robot uwbot-03, the driver simply reports:
Sep 20 14:24:36 uwbot-03 turtlebot4-start[4267]: [rplidar_composition-5] [INFO] [1726842276.821349066] [rplidar_composition]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version: '1.12.0'
Sep 20 14:24:36 uwbot-03 turtlebot4-start[4267]: [robot_state_publisher-7] [INFO] [1726842276.927562382] [robot_state_publisher]: got segment rplidar_link
Sep 20 14:24:41 uwbot-03 turtlebot4-start[4267]: [rplidar_composition-5] [ERROR] [1726842281.353785071] [rplidar_composition]: Error, operation time out. RESULT_OPERATION_TIMEOUT!
```

### To Reproduce

`ros2 topic echo /scan`

### Other notes

Aug 20 21:00:20 uwbot-03-base user.notice network_manager: ROBOT ID : 88157BA28E8949F1B90D26FEBE482A89
Sep 19 13:00:35 uwbot-02-base user.notice network_manager: ROBOT ID : 86E0D9F93734467A84EB9265755DABAE

hilary-luo, 2024-09-25:
@alexswerner since you are requesting support with an issue that needs troubleshooting please use the troubleshooting template. That way the issue will be assigned with the correct label and assignees right away. The bug template in this repo is only for when you find a software bug and can fully explain which section of code causes the bug behaviour and why.

alexswerner, 2024-09-26:
@hilary-luo fair enough, it was just not clear to me the "troubleshooting" includes hardware. Should I create a new ticket? Otherwise please let me know which specific information do you need here.

smatarCPR, 2024-09-27:
Hello @alexswerner
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4 unit. I have the following questions regarding the units behaviour:

- Typically, when the number of subscribers to the goes from a positive number to 0 the lidar will stop spinning. Initially there should be 0 subscribers on the lidar topic so it should not be spinning on startup, but it seems that the library use for the lidar has a bug that causes the lidar to begin spinning on startup.
- Out of curiosity, when you subscribe to the lidar topic on uwbot02 does the lidar start spinning? Are you able to visualize the lidar data in RVIZ on uwbot-02? 
- For uwbot-03, try disconnecting and reconnecting the lidar unit from the raspberry pi. The driver error reported seems to indicate a connection issue. If at all possible try swapping the lidar units from uwbot02 to uwbot03 and vise versa. See if the issues carry over from one unit to the next or if a different set of issues occurs

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

alexswerner, 2024-10-01:
Hi @smatarCPR !

* For uwbot-02 the lidar does start spinning when I subscribe to the `/scan` topic. It also stops correctly when the subscriber is stopped.
* For uwbot-03, I reconnected the lidar and the issue went away. Not that I did not already check the connection before, but at least this time this issue was fixed.

alexswerner, 2024-10-08:
Hi @smatarCPR,

how can we proceed with the broken lidar on uwbot-02. 

Thanks!

smatarCPR, 2024-10-11:
Hello @alexswerner 
In terms of uwbot-02's lidar, the behaviour seems within expectations. Were you able to visualise the data in RVIZ? If possible would you be able to provide a screenshot of the data being visualised in RVIZ. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-11-18:
Hello @alexswerner 
It's been quite some time since I've last heard from you. Do you have any updates you can share? Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-12-20:
Hello @alexswerner,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-12-20 by smatarCPR