turtlebot/turtlebot4 #478: COMM are constantly Off

CurtinRobotAI, 2024-09-24:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Not applicable

### Type of issue

Networking

### Expected behaviour

- Currently, we have a total of 10 Turtlebot4. Every two Turtlebot4s and two computers are connected to a Wifi Router (We have this setup because of frustration of constant failure communication between computers and the base of TB4s). 
- The communication between TB4s and computers should contain low latency considering current setting.
- Using simple discovery setting, computer can drive the robot by using "teleop_twist_keyboard" package, or by publishing a message through topic "/cmd_vel". 

### Actual behaviour

- Inconsistent behavior throughout the robot swarm, the communication between TB4 and computer for some robots achieved the expected behaviour, but constantly some robots and computers does not. 
- The time of establishing communication between TB and computers various, some are instantaneously but some take half of a hour.
- The communication latency between TB4 and computers increase over time. 

### To Reproduce

Not applicable.

### Other notes

We are using TurtleBot4 as our education platform. All students, professor, and technical support team are frustrated with this unreliable platform. Support is deadly needed.

smatarCPR, 2024-10-02:
Hello @CurtinRobotAI ,
Thank you for reaching out on the Turtlebot 4 Github. I am very sorry to hear that you are having issues with our products. Based on the current setup description I have the following recommendation:
Try using a discovery server setup with unique namespacing. Using a discovery server setup will reduce the network traffic and utilising unique namespaces will prevent topics and nodes from conflicting with each other.

Let me know how this recommendation works for you, if you have any follow up questions or if I can help with anything else.
Best Regards,
Saif M.

smatarCPR, 2024-10-30:
Hello @CurtinRobotAI ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-10-30 by smatarCPR