turtlebot/turtlebot4 #479: Error after switching Turtlebot to 5 GHz Wi-Fi

FishDentistry, 2024-10-01:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.0-1jammy.20231207.082431

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Navigation is operational and commands can be sent from RViz 

### Actual behaviour

After switching from a 2.4 GHz network to a 5 GHz network on the Turtlebot, Nav2 no longer responds to commands given in RViz, and is never displayed as active. A new error message also appears in the RViz logs. PC and Turtlebot are both using FastDDS

### Error messages

```bash
[RTPS_READER_HISTORY Error] Change payload size of '32' bytes is larger than the history payload size of '19' bytes and cannot be resized
```

### To Reproduce

1. Start SLAM, RViz, and Nav2 with PC and Turtlebot on 5GHz network with FastDDS configuration
2. Attempt to send command to Turtlebot
3. Nav2 does not respond, RViz outputs error as above

FishDentistry, 2024-10-10:
Update: After making the changes below, the localization and navigation errors are less frequent
1.  Within navigation_launch.py, I added the parameter '{'bond_timeout': 0.0}' to the lifecycle_manager node. 
2. Within the lifcycle_nodes list in the same file, I prepended all the node names with the namespace of my robot. Instead of reading 
`lifecycle_nodes = ['/sheldon/controller_server',
                       'smoother_server',
                       'planner_server',
                       '/behavior_server',
                       'bt_navigator',
                       'waypoint_follower',
                       'velocity_smoother']`
it now reads as
`lifecycle_nodes = ['/sheldon/controller_server',
                       '/robotnamespace/smoother_server',
                       '/robotnamespace/planner_server',
                       '/robotnamespace/behavior_server',
                       '/robotnamespace/bt_navigator',
                       '/robotnamespace/waypoint_follower',
                       '/robotnamespace/velocity_smoother']`
3. I performed the same prepending process for the lifecycle_nodes list in the nav2 localization_launch.py file, and added the same bond_timeout parameter to the lifecycle_manager node in this file as well
4.  In my nav2.yaml config file, I added 'bond_timeout: 0.0' under the ros__parameters of both bt_navigator and smoother_server

After doing the steps above, I now get navigation and localization working together with a namespace robot using the discovery server configuration. I don't know why this worked, and would appreciate insight to this, or suggestions on how to make these changes without editing those launch files directly. As for the RTPS error, the only partial solution I've found is to clear the /dev/shm directory on the robot and PC, and then reboot the robot.

smatarCPR, 2024-10-30:
Hello @FishDentistry ,
Thank you for reaching out on the Turtlebot 4 Github page. Apologies for the late reply on this. Would it be possible to provide the following items:

- May we please have the modified .py launch files you have added the bond_timeout parameter to
- Would it be possible to provide the full list of commands you are running in the order you are running them in
- Lastly, please provide the full terminal outputs of each command run.

Looking forward to hearing from you soon.
Best Regards,
Saif M.

smatarCPR, 2024-12-04:
Hello @FishDentistry 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-12-04 by smatarCPR

FishDentistry, 2024-12-10:
Apologies for my late response. At this moment, I have updated both the TB4 and my PC to the latest version of all packages. I no longer have any modifications in my config files or launch files. Localization and SLAM both work as expected, but navigation still constantly refuses to start. When starting it using the command "ros2 launch turtlebot4_navigation nav2.launch.py namespace:=robotnamespace" I get errors like these more than 95% of the time:

[controller_server-1] [WARN] [1733850699.044709052] [sheldon.controller_server.rclcpp]: failed to send response to /sheldon/controller_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314

[lifecycle_manager-8] [ERROR] [1733850769.066834196] [sheldon.lifecycle_manager_navigation]: Server controller_server was unable to be reached after 4.00s by bond. This server may be misconfigured.

[lifecycle_manager-8] [ERROR] [1733850769.066887196] [sheldon.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

[controller_server-1] [WARN] [1733850770.770257879] [sheldon.local_costmap.local_costmap]: Can't update static costmap layer, no map received

[lifecycle_manager-8] terminate called after throwing an instance of 'statemap::TransitionUndefinedException'

[lifecycle_manager-8]   what():  no such transition in current state

[ERROR] [lifecycle_manager-8]: process has died [pid 11076, exit code -6, cmd '/opt/ros/humble/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args --log-level info --ros-args -r __node:=lifecycle_manager_navigation -r __ns:=/sheldon --params-file /tmp/launch_params_xpcdo4n0 --params-file /tmp/launch_params_oa2_84m6 --params-file /tmp/launch_params_qlaz9wkq -r /sheldon/global_costmap/scan:=/sheldon/scan -r /sheldon/local_costmap/scan:=/sheldon/scan'].

[lifecycle_manager-8] [ERROR] [1733851742.518670284] [sheldon.lifecycle_manager_navigation]: Server bt_navigator was unable to be reached after 4.00s by bond. This server may be misconfigured.

[lifecycle_manager-8] [ERROR] [1733851742.518726090] [sheldon.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

[smoother_server-2] [WARN] [1733851944.251497818] [sheldon.smoother_server.rclcpp]: failed to send response to /sheldon/smoother_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314

[lifecycle_manager-8] [ERROR] [1733869538.939998241] [sheldon.lifecycle_manager_navigation]: Server controller_server was unable to be reached after 4.00s by bond. This server may be misconfigured.

[lifecycle_manager-8] [ERROR] [1733869538.940087550] [sheldon.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

[lifecycle_manager-8] terminate called after throwing an instance of 'statemap::TransitionUndefinedException'

[lifecycle_manager-8]   what():  no such transition in current state

I get these errors even when only launching navigation without anything else active