turtlebot/turtlebot4 #480: Ask for WiFi connection

Lynn-hh, 2024-10-04:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

22.04

### Type of issue

Networking

### Expected behaviour

After the boot, the wifi can be retrieved and successfully connected to the wifi

### Actual behaviour

Now I can only use the internet cable to ssh the turtlebot4. 
Then I check with it,  the first problem is the time error. It's still in 2023, but now is 2024.
When I use the "iwconfig" command to see the current status of the wireless card, it is not AP mode, it shows "managed". Maybe the current wireless NIC is in Managed mode
Then I disconnected the wifi connection, modified hostapd and started it. At this time, the wifi could be retrieved, but my computer could not connect to it.

### To Reproduce

iwconfig wlan0
the current wireless NIC is in Managed mode
sudo nmcli device disconnect wlan0
sudo nano /etc/hostapd/hostapd.conf
sudo systemctl restart hostapd

Lynn-hh, 2024-10-16:
> Hello @Lynn-hh <https://github.com/Lynn-hh> ,
> Thank you for reaching out on the Turtlebot 4 github. APologies for the
> late reply on this.
>
> After SSH'ing into the Turtlebot. are you able to make the necessary
> changes in the Turtlebot Setup Utility to connect to wi-fi (
> https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#connect-the-raspberry-pi-to-your-network)?
> To do so:
>
>    1. Run the command turtlebot4-setup
>    2. Scroll to Wi-Fi Settings and enter the W-fi settings sub menu.
>    3. Change Wi-Fi Mode from [Access Point] to [Client]
>    4. Enter the details of the network, SSID is the network name, the
>    option below that should be password.
>    5. Select the bandwidth of the wi-fi network: 5.0 GHz or 2.4 GHz.
>    6. Scroll down and select the Save option, you should be brought back
>    to the main menu. Scroll down and select the Apply Settings.
>    7. Exit the setup utility, the Raspberry Pi should reboot with the new
>    network settings applied.
>
> If the above process fails would it be possible to verify the following:
>
>    - Please provide the turtlebot logs using journalctl -b >
>    turtlebot4_log.txt
>    - Verify that the network you are connecting to does not have
>    additional security settings such as MAC address white listing or security
>    beyond a password.
>
> With regards to the date and time being out of sync, once the Turtlebot
> connects to a network it should sync. If it does not try the instructions
> found here to force a sync -
> https://ubuntu.com/server/docs/use-timedatectl-and-timesyncd
>
> Let me know how these recommendations work for you. Looking forward to
> hearing from you soon.
> Best Regards,
> Saif

Thank you so much, I will try what you suggest

smatarCPR, 2024-11-18:
Hello @Lynn-hh 
Were you able to follow the process outlined and were you successfully able to connect to wi-fi? Looking forward to hearing from you soon.
Best Regards,
Saif

Lynn-hh, 2024-11-18:
Yes！ After I refreshed the SD card, I can see the wifi of turtlebot4
normally. But I have a new problem, that is, our campus network requires
both username and password to log in. But the turtlebot4-set up given in
the tutorial only has a password to log in, and I can't enter the username.
What should I do?
Thank you so much for your help!
Lin

smatarCPR, 2024-11-20:
Hello Lin,
In this case I would do the following:

1. In the /etc/netplan folder create a yaml folder with a higher priority (i.e. lower number at the start of the file name) with the details of the network.
2. Follow the netplan documentation - https://netplan.readthedocs.io/en/stable to create the netplan yaml file for the network with a username and password
3. Following this run the command `sudo netplan apply`

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-01-20:
Hello Lin,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2025-01-20 by smatarCPR