turtlebot/turtlebot4 #483: Errors were encountered while processing: ros-humble-turtlebot4-setup

PrakritTyagi, 2024-10-14:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

```
ubuntu@ubuntu:~$ dpkg -s ros-$ROS_DISTRO-turtlebot4-setup
Package: ros-humble-turtlebot4-setup
Status: install ok half-configured
Priority: optional
Section: misc
Installed-Size: 158
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: arm64
Version: 1.0.4-1jammy.20240909.222719
Config-Version: 1.0.4-1jammy.20240729.103100
Depends: chrony, curl, libcurl4-openssl-dev, network-manager, ros-humble-rmw-cyclonedds-cpp, ros-humble-rmw-fastrtps-cpp, ros-humble-robot-upstart, ros-humble-simple-term-menu-vendor, ros-humble-ros-workspace
Conffiles:
 /etc/netplan/40-ethernets.yaml 6951b8a2f7fb3a647df50f19ed2b2762
 /etc/netplan/50-wifis.yaml e8658f314d8ccb4b2a1b5c4e4177dfa1
 /etc/systemd/system/webserver.service 2523ae6ccd96188391cb0cb7ea23c39b
 /etc/turtlebot4/aliases.bash e61a911906f05218fa606b6c5dfe2692
 /etc/turtlebot4/chrony.conf b7543efe4007b9480e0703dfccfce355
 /etc/turtlebot4/cyclonedds_rpi.xml 0ec64bd77cca8e8f6940486d2b02ae99
 /etc/turtlebot4/discovery.conf cbc9b4a66fec3957b98f258787587bd8
 /etc/turtlebot4/discovery.sh 8e476f2f051ffcd3f6062a9cefdb6d3e
 /etc/turtlebot4/fastdds_discovery_create3.xml 1abbfb9fd0db423dc7fdda2c083fdfbc
 /etc/turtlebot4/fastdds_rpi.xml 962330bbabde220a0eebee74d6f5b1ac
 /etc/turtlebot4/setup.bash d7b290c54b8eda578fb45dd021cad7d6
 /etc/turtlebot4/system a3b4be75d5acaf62cee13e9778ed9267
 /etc/udev/rules.d/50-turtlebot4.rules a9cb48367140746f5dc8bc0e37fd0c38
 /etc/udev/rules.d/80-movidius.rules 2f8a6174b951839e879fec25e7cdf624
 /etc/udev/rules.d/99-gpio.rules 9c389cf82c14c517d06232c2dac2599d
Description: Turtlebot4 setup scripts
```

### Type of issue

Select One

### Expected behaviour

I expect the that updating the TurtleBot 4 should finish without any errors.

### Actual behaviour

While updating the TurtleBot 4 ros-humble-turtlebot4-setup is not properly configured.

### Error messages

```bash
Installing new spi-rtc.dtbo.
Taking backup of spi2-1cs.dtbo.
Installing new spi2-1cs.dtbo.
Taking backup of cap1106.dtbo.
Installing new cap1106.dtbo.
Taking backup of w5500.dtbo.
Installing new w5500.dtbo.
Taking backup of minipitft13.dtbo.
Installing new minipitft13.dtbo.
Taking backup of README.
Installing new README.
Errors were encountered while processing:
 ros-humble-turtlebot4-setup
needrestart is being skipped since dpkg has failed
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### To Reproduce

1. Flashed the sd-card using 
``` wget https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/scripts/sd_flash.sh bash sd_flash.sh ~/Downloads/turtlebot4_lite_humble_1.0.4.img ```
2. SSH into the Raspberry Pi using 
```ssh ubuntu@10.42.0.1```
3. Connected the Raspberry Pi to my network by following the instructions after
```turtlebot4-setup```
4. Found the new IP address and SSH into it.
5. Updating the TurtleBot 4
```sudo apt update && sudo apt upgrade```

### Other notes

Note: 

- This Turtlebot 4 was previously being used in another project. I wanted to start from scratch so followed the Basic Setup page.
- This is how the App Config page on Create 3 Webserver looks like.
[image]
- I found that out of all the necessary Conffiles needed for turtlebot4-setup only one is missing in the Operating system :
/etc/turtlebot4/chrony.conf
- I tried installing ros-$ROS_DISTRO-turtlebot4-setup separately but it failed as it cannot mv the missing file.
```ubuntu@ubuntu:~$ sudo apt install ros-$ROS_DISTRO-turtlebot4-setup
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-humble-turtlebot4-setup is already the newest version (1.0.4-1jammy.20240909.222719).
The following packages were automatically installed and are no longer required:
  linux-headers-5.15.0-1059-raspi linux-image-5.15.0-1059-raspi linux-modules-5.15.0-1059-raspi linux-raspi-headers-5.15.0-1059
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up ros-humble-turtlebot4-setup (1.0.4-1jammy.20240909.222719) ...
mv: cannot stat '/etc/turtlebot4/chrony.conf': No such file or directory
dpkg: error processing package ros-humble-turtlebot4-setup (--configure):
 installed ros-humble-turtlebot4-setup package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 ros-humble-turtlebot4-setup
needrestart is being skipped since dpkg has failed
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
- Fulle log file after running
```sudo apt update && sudo apt upgrade```
[error.log](https://github.com/user-attachments/files/17357963/error.log)

PrakritTyagi, 2024-10-14:
I am thinking of creating the chrony.conf file and copying the contents of the file from [turtlebot4_setup](https://github.com/turtlebot/turtlebot4_setup/tree/humble) package.

smatarCPR, 2024-10-18:
Hello @PrakritTyagi 
Thank you for reaching out on the Turtlebot 4 Github page. A bit of background on this matter, there is a known bug with regards to obtaining the chrony.conf file, but we have pushed a bug fix for this, but copying the file template and creating it from scratch is a viable option. From our end it the file appears fine and the terminal output seems reasonable. Are you encountering any operational issues with the Turtlebot?

In terms of the kernel, where is this update coming from? Have you connected a monitor to the Raspberry Pi?

Looking forward to hearing from you soon.
Best Regards,
Saif

PrakritTyagi, 2024-10-21:
HI @smatarCPR 
Thanks for replying. I was waiting for the reply, and I have not tried out operating the Turtlebot with turtlebot4_lite_humble_1.0.4.img, ill let you know.

In terms of kernel I get the kernel update msg through ssh. I didnt connect the monitor to the Raspi.

ill do everything again and if i encounter any operational issue ill let you know.
Best Regards.

smatarCPR, 2024-10-30:
Hello @PrakritTyagi ,
Apologies for the late reply on this. We don't not have a specified kernel version, you are free to use the most up to date version of the kernel available to you. The kernel screen you are seeing is simply a notification that a newer version of the kernel is available for use and you should restart the Pi so that the kernel can be used.

Are you able to get past this kernel screen?

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-12-04:
Hello @PrakritTyagi ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

my-name-is-D, 2025-01-13:
Hello, is there any update? I have the same issue in 2025 downloading from http://download.ros.org/downloads/turtlebot4/ and following the same steps. 

My version is different, but that's it:
Version: 1.0.4-1jammy.20241128.023135
Config-Version: 1.0.4-1jammy.20240729.103100

smatarCPR, 2025-01-22:
Hello @PrakritTyagi and @my-name-is-D ,
This specific bug has been resolved in version 1.0.5 which will released on the next synchronization. Please update when you can.
I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2025-01-22 by smatarCPR