turtlebot/turtlebot4 #486: LiDAR data can not be seen from User PC

myamo1, 2024-10-27:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

[image]

### Type of issue

LIDAR

### Expected behaviour

Subscribe from User PC to /scan topic and see scan msg.

### Actual behaviour

Nothing appears (I can not see the /scan msg)

### To Reproduce

Run **ros2 topic echo /scan** on User PC.

### Other notes

- The LiDAR is turning fine and I can see the messages when I run **ros2 topic echo /scan** from the turtlebot.
- Here is the output when I run ros2 topic list from the turtlebot:
[image]
- Here is the output when I run ros2 topic list from the User PC:
[image]
- It was working fine, and then I started having this issue occasionally, but now it is persistent.

smatarCPR, 2024-11-06:
Hello @myamo1 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with one of our products. I have a couple of questions regarding the current functionality of the Turtlebot:

- When running the echo command, is the Turtlebot docked or in an undocked state?
- If you echo the /scan topic when SSH'ed into the Raspberry Pi on the Turtlebot is data output?
- Besides the /scan topic have you noticed other topics that you are unable to echo on the user computer?
- Is the Lidar spinning when you begin echoing the /scan topic?

Looking forward to hearing from you soon.
Best Regards,
Saif

myamo1, 2024-11-07:
Hello @smatarCPR,
Thanks for your reply. 

- I tried In both scenarios (docked and undocked) but I get the same result.
- Yes, when SSH'ed into the RPi on the Turtlebot and echo the /scan topic, the data is outputed.
- As I mostly work with the /scan and /odom topics, I have not tried other topics but something similar happened only once or twice with the /odom topic. Frim since, only the /scan topic is not discoverable from my PC.
- Yes, the Lidar is spinning when I begin echoin the /scan topic.

I observed that this issue only happens when the topics shown from my PC, when running 'ros2 topic list', are not exactly the ones shown from the RPi on the Turtlebot as you can see from the two previously posted screenshots.

Regards,

Ishak

smatarCPR, 2024-11-18:
Hello @myamo1 
Thank you for providing these details. I have the following follow up questions regarding your current setup:

- Are you currently using a 5 GHz network for the Raspberry Pi? How is the network you are using setup? This may be due to a networking issue.
- Does the issue persist if you switch over to a discovery server setup? 

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-12-20:
Hello @myamo1 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-12-20 by smatarCPR