turtlebot/turtlebot4 #488: turtblebot4 not working LED lights off and cannot connect

Torreskai0722, 2024-10-28:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

all default packages

### Type of issue

Other

### Expected behaviour

works as the manual to connect.

### Actual behaviour

The device cannot connect and middle three lights are off. I try to use ethernet cable to connect and reset the networking configurations. But it disconnects very shortly.
[image]

### To Reproduce

Just as descirbed.

RustyCPR, 2024-10-30:
Hi @Torreskai0722 
What is the Create 3 ring light status? https://iroboteducation.github.io/create3_docs/hw/face/
The COMM led off indicating that the connection between the RPI4 and Create3 is not up.

Torreskai0722, 2024-11-07:
Create 3 Ring light is solid white.

I apply default setting to tuetlebot4 and now only two LED lights off (COMM, BATT).

How to address the connection problem?

> Hi @Torreskai0722 What is the Create 3 ring light status? https://iroboteducation.github.io/create3_docs/hw/face/ The COMM led off indicating that the connection between the RPI4 and Create3 is not up.

RustyCPR, 2024-11-07:
Thanks @Torreskai0722 
Please verify the Create3 to PCBA connections are securely connected see: https://github.com/turtlebot/turtlebot4/issues/394#issuecomment-2045931389

Can you please try connecting to a WiFi network - detailed in the [user manual. ](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html)

Thank you,
Rusty

Torreskai0722, 2024-11-08:
> #394 (comment)

I checked the connection between Create3 to PCBA, which is good. But the LED lights for COMM and BATT are still off.

RustyCPR, 2024-11-13:
Hi @Torreskai0722 
Thanks for the feedback. 
Can you please try commenting out this [ line](https://github.com/turtlebot/turtlebot4_setup/blob/6e8a21057c0074576268c2efc6135f0f53124745/etc/netplan/40-ethernets.yaml#L8) in your PC's Netplan ethernet yaml config file - this should allow you to connect without it disconnecting. 

Please try running the [Setup Tool](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools) and re-establish communication

Please let me know how it goes? 
Regards,
Rusty

RustyCPR, 2024-12-20:
Any feedback on this issue @Torreskai0722
Is this issue resolved or still on going? 
Thanks,
Rusty