turtlebot/turtlebot4 #489: /stop_motor Service Does Not Stop the RP LIDAR Motor

aaronwilhelm1, 2024-10-31:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.5-1jammy.20240731.002306 for ros-humble-turtlebot4-node package 

### Type of issue

LIDAR

### Expected behaviour

When running the command ```ros2 service call /stop_motor std_srvs/srv/Empty``` I expect the LIDAR to stop spinning.

### Actual behaviour

When running the command ```ros2 service call /stop_motor std_srvs/srv/Empty``` the LIDAR does _not_ stop spinning.

### Error messages

```bash
ubuntu@ubuntu:~$ ros2 service call /stop_motor std_srvs/srv/Empty
waiting for service to become available...
requester: making request: std_srvs.srv.Empty_Request()

response:
std_srvs.srv.Empty_Response()
```

### To Reproduce

1. Run ros2 service call /stop_motor std_srvs/srv/Empty

### Other notes

I am doing this from an SSH session into the Raspberry Pi on the Turtlebot 4 Lite. I have tried it while the robot is both on the dock and off the dock. The robot is responsive in other ways, for instance I can still drive around the Turtlebot 4 using a controller

RustyCPR, 2024-11-06:
Hi good afternoon @aaronwilhelm1 
Can you please try this suggestion: https://github.com/turtlebot/turtlebot4/issues/405#issuecomment-2083737967

Let me know,
Rusty

aaronwilhelm1, 2024-11-09:
Manually installing it didn't help unfortunately, but on the page [here](https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#controlling) it notes a bug:

> There is currently a bug which results in the scan topic only stopping when the subscriber count drops from 1 or more to 0. This means that the motor will run until the scan topic is subscribed to and then unsubscribed from for the first time.

That's exactly what I'm running into. By just listening to the /scan topic from the command line and then stopping I'm able to get the lidar to stop spinning. Thanks for the help!

closed 2024-11-09 by aaronwilhelm1