turtlebot/turtlebot4 #490: Can't drive TurtleBot4

kylechen357, 2024-11-02:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.0-1jammy.20240830.234728

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I want to use teleop_twist_keyboard to drive my TurtleBot 4

### Actual behaviour

When I click the keyboard, nothing happened.
User PC ros2 topics:
/battery_state
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/rgb/preview/image_raw
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
TurtleBot 4 ros2 topics:
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/hazard_detection
/hmi/buttons
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ir_intensity
/ir_opcode
/joint_states
/joy
/kidnap_status
/mobility_monitor/transition_event
/mouse
/oakd/rgb/preview/image_raw
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

### To Reproduce

~

smatarCPR, 2024-11-06:
Hello @kylechen357 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with one of our products. I have a couple of questions regarding the state of the Turtlebot:

- Is the Turtlebot able to otherwise drive using either cmd_vel or the bluetooth controller?
- Is the Turtlebot docked or undocked?
- What command are you using to enable keyboard teleop, where are you running this command?
- If the Turtlebot is unable to drive regardless of the method used would it be possible to share the [Create3 logs?](https://iroboteducation.github.io/create3_docs/webserver/logs/)

Looking forward to hearing from you soon.
Best Regards,
Saif

kylechen357, 2024-11-07:
@smatarCPR Thanks for the response, here to answer some questions you ask:
1. When we bought the TurtleBot 4 lite, it doesn't give us the Bluetooth controller.
2. Either TurtleBot undocked or docked were both tested, they all don't get any response.
3. I were using the command of 'ros2 run teleop_twist_keyboard teleop_twist_keyboard' to enable keyboard teleop.
4. The log file is attached in this comment, and glad to hear the response from you.
Thanks.
[messages.txt](https://github.com/user-attachments/files/17655395/messages.txt)

kylechen357, 2024-11-08:
Dear @smatarCPR ,
Thanks for the response. And I have tested the command above that you mentioned in both user PC and TB4 environment.
Both run the error:
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.11', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/pub.py", line 222, in main
    return main(args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/pub.py", line 240, in main
    return publisher(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/pub.py", line 271, in publisher
    values_dictionary = yaml.safe_load(values)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 577, in fetch_value
    raise ScannerError(None, None,
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 1, column 11:
     linear: x: 0.5 y: 0.0 z: 0.0 angular: x:  ..
Moreover, I hope to hear the solution soon. Thank you.

smatarCPR, 2024-11-18:
Hello @kylechen357 
I have a couple questions/recommendations I wish for you to try:

- For the cmd_vel command there is an extra slash, typically used when the command is broken into multiple lines, try this form of the command - `ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "linear: x: 0.5 y: 0.0 z: 0.0 angular: x: 0.0 y: 0.0 z: 0.0"`
- Is there communication between the Create3 and your computer (try echoing the /odom topic and seeing if you are getting the data)
- After publishing to cmd_vel, try subscribing to cmd_vel in a different terminal and seeing if the command you sent is indeed being published.
- One last recommendation would be to ensure that your packages have been updated per - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#updating-the-turtlebot-4

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-12-20:
Hello @kylechen357 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-12-20 by smatarCPR