turtlebot/turtlebot4 #493: Not seeing ros2 topics on pc

rohanchandra-cpe, 2024-11-10:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I'm not sure. I'm using the following docker image, which has ros2 humble installed already: https://github.com/Tiryoh/docker-ros2-desktop-vnc

### Type of issue

Networking

### Expected behaviour

I'm able to see all of the turtlebot4 and create3 topics on both my pc and on the raspberry pi

### Actual behaviour

I'm able to see the turtlebot4 and create3 topics on the raspberry pi, but not on my pc

### Error messages

```bash
No error messages.
```

### To Reproduce

1. Run `ros2 topic list` in the terminal on my pc (within the docker image)
2. The topics will not appear

PrakritTyagi, 2024-11-14:
Hi could you try running the docker image with also these two arguments --privileged --network=host and see if you can see topics in the container?

rohanchandra-cpe, 2024-11-16:
@PrakritTyagi  This is what happens when I try to run it with those arguments:

2024-11-15 21:48:46 * enable custom user: ubuntu
2024-11-15 21:48:46 useradd: warning: the home directory /home/ubuntu already exists.
2024-11-15 21:48:46 useradd: Not copying any file from skel directory into it.
2024-11-15 21:48:46   set default password to "ubuntu"
2024-11-15 21:48:46 grep: /home/ubuntu/.bashrc: No such file or directory
2024-11-15 21:48:46 ============================================================================================
2024-11-15 21:48:46 NOTE 1: --security-opt seccomp=unconfined flag is required to launch Ubuntu Jammy based image.
2024-11-15 21:48:46 See https://github.com/Tiryoh/docker-ros2-desktop-vnc/pull/56
2024-11-15 21:48:46 ============================================================================================
2024-11-15 21:48:46 2024-11-16 02:48:46,429 INFO Included extra file "/etc/supervisor/conf.d/supervisord.conf" during parsing
2024-11-15 21:48:46 2024-11-16 02:48:46,430 INFO Set uid to user 0 succeeded
2024-11-15 21:48:46 2024-11-16 02:48:46,431 INFO RPC interface 'supervisor' initialized
2024-11-15 21:48:46 2024-11-16 02:48:46,431 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2024-11-15 21:48:46 2024-11-16 02:48:46,431 INFO supervisord started with pid 43
2024-11-15 21:48:47 2024-11-16 02:48:47,440 INFO spawned: 'novnc' with pid 44
2024-11-15 21:48:47 2024-11-16 02:48:47,444 INFO spawned: 'vnc' with pid 45
2024-11-15 21:48:47 2024-11-16 02:48:47,560 INFO exited: novnc (exit status 1; not expected)
2024-11-15 21:48:48 2024-11-16 02:48:48,633 INFO spawned: 'novnc' with pid 632
2024-11-15 21:48:48 2024-11-16 02:48:48,633 INFO success: vnc entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2024-11-15 21:48:48 2024-11-16 02:48:48,708 INFO exited: novnc (exit status 1; not expected)
2024-11-15 21:48:50 2024-11-16 02:48:50,715 INFO spawned: 'novnc' with pid 681
2024-11-15 21:48:50 2024-11-16 02:48:50,817 INFO exited: novnc (exit status 1; not expected)
2024-11-15 21:48:53 2024-11-16 02:48:53,836 INFO spawned: 'novnc' with pid 703
2024-11-15 21:48:53 2024-11-16 02:48:53,946 INFO exited: novnc (exit status 1; not expected)
2024-11-15 21:48:54 2024-11-16 02:48:54,953 INFO gave up: novnc entered FATAL state, too many start retries too quickly

RustyCPR, 2024-11-18:
Hi there @rohanchandra-cpe 
Can you please make sure you have gone through the setup process and installed all required TB4 packages on the remote PC? [LINK](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#user-pc)

Can you try verifying basic networking, ensure that you can communicate between the docker container and other outside computers (ping, ssh etc.) and after that is fully functional, then test ROS networking and then test turtlebot4. 

It appears you may have a configuration issue - if possible try running on a standard Ubuntu 22.04 ROS Humble docker image and make sure it works then move to the custom container image you are working with? 
We don't directly support containerized systems because of the complexity and uniqueness of different configurations. 

There are various resources available including Robotics Stack Exchange (https://robotics.stackexchange.com/) and the Open Robotics Discord.

Best regards,
Rusy

RustyCPR, 2024-12-20:
Closing due to no reply or activty.

closed 2025-01-27 by RustyCPR