turtlebot/turtlebot4 #494: Can't teleoperate TB4 with remote PC

ValentinLeglise, 2024-11-12:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

*

### Type of issue

Networking

### Expected behaviour

After connected the robot to the remote PC, the robot of my customer should be able to move after running "ros2 run teleop_twist_keyboard teleop_twist_keyboard", and should see all the TB4 topics with the command "ros2 topic list".

### Actual behaviour

My customer can't drive his robot remotely with his PC. When he uses "ros2 run teleop_twist_keyboard teleop_twist_keyboard", or "ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0" " , the robot doesn't respond. But, when he is in SSH into the robot, "ros2 topic pub /cmd_vel" works, but "ros2 run teleop_twist_keyboard" doesn't.

When he runs "ros2 topic list" on its PC, my customer only see 2 topic /rosout and /parameter_events.

Its probably a sourcing issue but, his .bashrc file and /etc/turtlebot4/setup.bash seems to be good:

.bashrc file:

'# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color|*-256color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
        # We have color support; assume it's compliant with Ecma-48
        # (ISO/IEC-6429). (Lack of such support is extremely rare, and such
        # a case would tend to support setf rather than setaf.)
        color_prompt=yes
    else
        color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# colored GCC warnings and errors
#export GCC_COLORS='error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01'

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi

CYCLONEDDS_URI='<CycloneDDS><Domain><General><DontRoute>true</></></></>'
export CYCLONEDDS_URI='<CycloneDDS><Domain><General><DontRoute>true</></></></>'
source /etc/turtlebot4/setup.bash
source /etc/turtlebot4_discovery/setup.bash
#export ROS_DOMAIN_ID=30 #TURTLEBOT3'

/etc/turtlebot4/setup.bash

$ cat /etc/turtlebot4/setup.bash
source /opt/ros/humble/setup.bash
export ROS_DOMAIN=0
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp

### To Reproduce

1. Connect Tb4 to remote PC
2. run "ros2 run teleop_twist_keyboard teleop_twist_keyboard"

PrakritTyagi, 2024-11-22:
Hi, First thing I noticed in your bashrc file is you are sourcing two setup.bash. One is for simple discovery and other is for discovery server. Comment out "source /etc/turtlebot4_discovery/setup.bash " and restart your ros2 daemon using

> ros2 daemson stop 

>  ros2 daemon start

ValentinLeglise, 2024-11-25:
Hello @PrakritTyagi,

Thank you for your advice, I will suggest this to my customer.

ValentinLeglise, 2024-11-29:
Hello,

My customer modified his bashrc file and now, in SSH he sees all topics of the robot:

    ubuntu@ubuntu:~$ ros2 topic list
    /battery_state
    /cliff_intensity
    /cmd_audio
    /cmd_lightring
    /cmd_vel
    /diagnostics
    /diagnostics_agg
    /diagnostics_toplevel_state
    /dock_status
    /function_calls
    /hazard_detection
    /imu
    /interface_buttons
    /ip
    /ir_intensity
    /ir_opcode
    /joint_states
    /joy
    /joy/set_feedback
    /kidnap_status
    /mobility_monitor/transition_event
    /mouse
    /oakd/rgb/preview/image_raw
    /odom
    /parameter_events
    /robot_description
    /robot_state/transition_event
    /rosout
    /scan
    /slip_status
    /static_transform/transition_event
    /stop_status
    /tf
    /tf_static
    /wheel_status
    /wheel_ticks
    /wheel_vels

But the output of same command in his PC machine is:

    $ ros2 topic list
    /parameter_events
    /rosout

What can he do to correct this? Is it really a sourcing problem?

Thanks in advance for your help.

ValentinLeglise, 2024-12-02:
Hello ClearPath.

Is it possible to have a response for this ?

pvg6, 2024-12-07:
Hi @aaronchongth is that no need to modify any XML file for the cyclonedds_uri..

aaronchongth, 2024-12-07:
Hey @pvg6, yeah in my case, I just left `CYCLONEDDS_URI` empty so that it will just use the default config

pvg6, 2024-12-07:
@aaronchongth 
I followed this [https://hub.docker.com/r/theconstructai/zenoh-bridge-ros2dds](url)

PI to Create3 communication is but host to PI is not there how can i connect to PI..
or better to switch discovery server

aaronchongth, 2024-12-07:
@pvg6 yup that will only resolve the connection between the Create3 and Raspberry Pi, but not the host because it bridges domain 0 and 1 internally.

My method sets both the Create3 and the Raspberry pi to use domain 0 and cyclonedds with default configurations. This should already allow SSH into the raspberry pi to teleop, dock/undock. Then set up bridges between the raspberry pi and host PC, so we can teleop from host PC without SSH.

pvg6, 2024-12-08:
@aaronchongth the solution you provided will solve the memory issues ( Full red light) caused by create3..
[https://hub.docker.com/r/theconstructai/zenoh-bridge-ros2dds](url) this solved  the memory issue..

please how can i solve the comple red light issue...

aaronchongth, 2024-12-08:
@pvg6 according to https://iroboteducation.github.io/create3_docs/hw/face/, solid red light ring means a robot error and not related to networking. Looks like you will need to cycle power, or debug further

LotfiZ, 2025-02-25:
@aaronchongth is it the only way to be able to use cycloneDDS between a TB4 and a remote machine ? This is quite strange as in the documentation they never mention that! In addition of that, it works with fast rtps. Do you have any idea on this ?

aaronchongth, 2025-02-26:
hey @LotfiZ, it's definitely not the only way, but I found this way to be fuss free and worked well, especially since I was already playing around with zenoh. The official documentation does not cover this as I only found it out through experimentation, and have not found a place to contribute upstream other than answering in issues.

aaronchongth, 2025-02-27:
Cool to know! Since this method is rather unorthodox and different from the main tutorial, I believe folks handling the tb4 will need to be in the loop of whether this method should be incorporated. Any thoughts @RustyCPR?

RustyCPR, 2025-04-14:
Hi @ValentinLeglise 
Appologies for the delayed reply - just saw this. 
Lots of discussion here - what's the current status of the issue? 
Thank you,
Rusty

aaronchongth, 2025-04-15:
Hi @RustyCPR! I might be mistaken, but the TLDR seems to be that folks were having issues connecting a remote host to a TB4, I suggested my own funky way of using Zenoh bridges, judging from the follow-up, seem to have resolved their issues...?

I basically set Tb4 and Create to CycloneDDS, removing all custom DDS configs (use default ones), and spin up Zenoh bridges on both the Tb4 and my remote PC (a Zenoh router on the remote PC helps a lot too). The Zenoh bridge on the Tb4's pi can pick up all the DDS messages from both the Tb4 and the Create, which allows my remote PC to teleop as well as start the nav stack. Not sure if this is a connectivity method that the team wants to explore and perhaps have on their official documentation