turtlebot/turtlebot4 #496: ROS2 Jazzy Bluetooth doesn't work

UdayWaveMark, 2024-11-14:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

Jazzy

### Type of issue

Other

### Expected behaviour

Should be able to connect the Controller with Turtlebot4 without any issues

### Actual behaviour

Short version: Bluetooth DOESN'T WORK.  (Jazzy, Ubuntu 24.04)

Long version:
Since this form doesn't have an option to select the following I have selected the existing options for ROS Distro and OS.  

ROS Distro: Jazzy
OS: Ubuntu 24.04

First I would like to say that I have experimented with ROS2 Humble and setup my Turtlebot4 Standard and was able to use RViz, create a map of a location and managed to do autonomous driving of that map.

My issue is with ROS2 Jazzy.  Created Jazzy image, replaced the old SD card (that had Humble) with the Jazzy image, followed the Basic Setup instructions + Discovery Server setup and finished setting up the Turtlebot4 with Jazzy.

Then as it is mentioned in the 'Supplemental Configuration', to connect my controller, I SSH into the Turtlebot4 and follow the instructions nothing works.

_Please see the 'Error messages' section for the commands that I have issued and the errors I received for those commands._

**Simply put, NOTHING to do with Bluetooth seems to WORK or even EXIST.**

### Error messages

```bash
sudo bluetoothctl
[sudo] password for ubuntu: 
sudo: bluetoothctl: command not found.

ubuntu@turtlebot4:~$ sudo bluetooth.sh
[sudo] password for ubuntu: 
sudo: bluetooth.sh: command not found.

ubuntu@turtlebot4:~$ systemctl status bluetooth
Unit bluetooth.service could not be found.

ubuntu@turtlebot4:~$ systemctl status bluetooth.service
Unit bluetooth.service could not be found.

ubuntu@turtlebot4:~$ systemctl start bluetooth.service
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ====
Authentication is required to start 'bluetooth.service'.
Authenticating as: ubuntu
Password: 
==== AUTHENTICATION COMPLETE ====
Failed to start bluetooth.service: Unit bluetooth.service not found.
```

### To Reproduce

sudo bluetoothctl

sudo bluetooth.sh

UdayWaveMark, 2024-11-14:
Also I tried the 'turtlebot4_setup' (ros2 run turtlebot4_setup turtlebot4_setup) tool and 'Bluetooth Setup' option.  

That too does nothing.

smatarCPR, 2024-11-20:
Hello @UdayWaveMark ,
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry to hear that you are having issues with one of our products. I wanted to bring to your attention a bug fix that was recently released for the Jazzy installation. The new updated ISO is version 2.0.1. Is this the version of the ISO used for the Raspberry Pi? Let me know if this is the case.

Looking forward to hearing from you soon.
Best Regards,
Saif

UdayWaveMark, 2024-11-25:
@smatarCPR ,

As your suggestion, I downloaded the latest ISO version 2.0.1 of Jazzy and created a new image and used that.  All the problems I mentioned seems to be resolved.  All the bluetooth related files/libraries are available.  Also when issue 'ros2 topic list' from the PC, it seems to be displaying all the topics.

Thanks very much for your help.

Uday

closed 2024-11-25 by UdayWaveMark