turtlebot/turtlebot4 #497: is there a way to have uncompressed depth image topic?

IacopoFornai, 2024-11-19:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Select One

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

#depth

### Type of issue

Camera

### Expected behaviour

i would like to have depth-image in an uncompressed format to run visual odometry (rtabmap), 

### Actual behaviour

so far i have only compressed depth image topic (/turtlebot4_4/oakd/rgb/preview/image_raw/compressedDepth), 

### To Reproduce

 ros2 topic list

RustyCPR, 2024-11-27:
Hi there, 
Depth topics are not enabled by default. 
To enable the OAK-D depth functionality please follow this [response](https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838).
This is one of the areas where we would like to create manual tutorials - if you find a process that works for you - please let us know. 
Thanks,
Rusty

RustyCPR, 2025-01-27:
Closing due to inactivity.

closed 2025-01-27 by RustyCPR