turtlebot/turtlebot4 #498: Create discovery server partitions on User PC for multiple turtlebots

PrakritTyagi, 2024-11-19:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

turtlebot4_standard_humble_1.0.0.zip	2023-03-03 23:51	1.9G

### Type of issue

Networking

### Expected behaviour

Connect individual terminals to each turtlebots.

### Actual behaviour

Each terminal opened becomes a listener to all turtlebots. 

### To Reproduce

1. open individual terminals
2. Soure each ROS_DISCOVERY_SERVER address:port in each terminal 
3. Listen to each turtlebot topics in individual terminals

### Other notes

Hi, I have four turtlebots setup with discovery server. The discover server setup that configures the user PC has created a setup.bash file in the directory /etc/turtlebot_discovery. The setup.bash file looks like this...

```
source /opt/ros/humble/setup.bash
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
[ -t 0 ] && export ROS_SUPER_CLIENT=True || export ROS_SUPER_CLIENT=False
export ROS_DOMAIN_ID=0
export ROS_DISCOVERY_SERVER="192.168.0.106:11811;192.168.0.41:11811;192.168.0.151:11811;192.168.0.112:11811;"
```

I want to change my setup so that I can listen to ros topics/services/actions for any robot when I export its discovery server address:port. Changes that I have in my mind:

1. Setup discovery server using different port for each robot as explained in the [Ros2 humble docs](https://docs.ros.org/en/humble/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#discovery-partitions) .
2. Create individual setup.bash files for each robot that only exports their server address. How should the content in these bash files look like?
3. Can someone explain each line in my current setup.bash file? I want to understand what happens when I source the above bash file.

I would greatly appreciate if someone would look at this and give me suggestions to get this working. Thank you for your guidance.

hilary-luo, 2024-12-11:
Apologies for the delay in my response. You have the right ideas overall but there are some caveats.

First I want to address an issue with the core request. It is definitely possible to set up the computer to listen only to one particular robot at a time, however the ROS 2 CLI tools rely on something called the ROS 2 Daemon. There is only one daemon and that daemon can only run one set of communication settings at a time.  For more information on what the ROS 2 daemon is read this draft documentation that will be going up on the clearpath documentation website soon: https://github.com/clearpathrobotics/cpr-documentation/blob/feature/networking/docs_versioned_docs/version-ros2humble/ros/networking/ros2_communication.mdx 

So this means that although setup.bash files can be created for each robot, the ROS 2 CLI cannot be used with more than one robot at a time and when switching between robots the daemon would need to be restarted. If you are not planning to use the CLI at all in these individual terminals then using the individual setup files for each terminal will be sufficient. If you want to use the CLI then a different solution is necessary. Depending on what you are trying to do it may be more useful to use some other type of filtering when running commands in different terminals while still having the daemon connected to all robots. Alternatively, you can ssh into each robot and run the commands on board. By default the robots will not be able to see each other and functionally the result of using the CLI would be similar. I would need more context to provide more concrete recommendations. I will still answer your specific questions below though:

> 1. Setup discovery server again and use different port for each robot as explained in the [Ros2 humble docs](https://docs.ros.org/en/humble/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#discovery-partitions) .

You only need to set a different port for different discovery servers if you are running multiple servers on the same computer. If you are still only going to be running the one server on each robot then no changes are necessary here. In that example the ports have to be unique because two servers are run on the same computer. If you still want to change the ports, it can be done through the Turtlebot4 setup tool under discovery server.

> 2. Create individual setup.bash files for each robot that only exports their server address. How should the content in these bash files look like?

Assuming that you are leaving the discovery servers on the default port, all you need to change is to remove the extra IP's for the robots that you do not want to listen to while keeping any of the semicolons. These semicolons designate the server ID. Use the same setup.bash file except change the `ROS_DISCOVER_SERVER` export line to one of the following:

For robot 1 use:
```bash
export ROS_DISCOVERY_SERVER="192.168.0.106:11811"
```
For robot 2 use:
```bash
export ROS_DISCOVERY_SERVER=";192.168.0.41:11811"
```
For robot 3 use:
```bash
export ROS_DISCOVERY_SERVER=";;192.168.0.151:11811"
```
For robot 4 use:
```bash
export ROS_DISCOVERY_SERVER=";;;192.168.0.112:11811;"
```

> 3. Can someone explain each line in my current setup.bash file? I want to understand what happens when I source the above bash file.

```bash
source /opt/ros/humble/setup.bash # This tells the terminal where to find the basic ROS commands and defines which of the installed ROS distributions is used
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp # This chooses the ROS Middleware, which vendor's software is used to manage ROS 2 messages, Fast DDS or Cyclone DDS
[ -t 0 ] && export ROS_SUPER_CLIENT=True || export ROS_SUPER_CLIENT=False # This checks if it is being run in the terminal and if so, it sets the terminal as super client. That means that the terminal will have access to all information from the discovery servers, not just what it knows about in advance (necessary for ROS CLI but undesired for normal nodes)
export ROS_DOMAIN_ID=0 # Sets domain Id but that doesn't matter for discovery server
export ROS_DISCOVERY_SERVER="192.168.0.106:11811;192.168.0.41:11811;192.168.0.151:11811;192.168.0.112:11811;" # Sets which servers to connect to and indicates the IP address, Server ID and Port number for each of them.
```

PrakritTyagi, 2024-12-16:
Thank you for the clear and thorough explanation—it answered all of my questions perfectly.

I’ve been encountering some discovery issues when running ROS 2 nodes and services. In my setup, I have four Turtlebots, each using a discovery server, and I run AMCL and Nav2 on these Turtlebots by SSHing into them from my laptop. As a result, I need to open eight SSH terminals plus one more terminal to run a navigator node that sends goal locations to each Turtlebot. Even though I can see the services with the ROS 2 command-line tools, the navigator node sometimes reports that it’s still waiting for `waitUntilNav2Active` on the third or fourth Turtlebot.

Because of these issues, I’m considering separating the discovery server to help reduce network clutter and simplify the process of locating topics and services. 

Any other suggestion would be greatly appreciated.

hilary-luo, 2024-12-20:
I need to better understand the situation a bit before I can provide much in terms of suggestions but here are my thoughts and questions so far.

In regards to opening so many terminals, you could set up Nav2 and localization to launch automatically on the robots so that you don't have to launch them manually each time. (Create a new Ubuntu service that runs a script to start the nodes, you could put them together into 1 launch file that gets started by the script. Then you can start / stop / enable / disable the service as desired.)

My understanding from what you have said is that each Turtlebot has been set up with it's own Discovery Server. Do you have these discovery servers connected? (aka can turtlebot #1 see the topics from turtlebot #2 and turtlebot #3 etc?). Based on what you have said about running into issues by robot 3 and 4, it sounds like you could have them connected or it could be the issue I talk about in the last paragraph.

What do you mean by "separating the discovery server". I would suggest not having the servers connected. That being said, by default they should not be connected.

What I suspect may be part of the issue here is the role of super clients. By default, your terminal (on the robot and offboard) will act as a super client. This is required for the CLI tools to work properly (to be able to run topic list and node list etc.). However, this can be problematic when you are launching large nodes directly from terminal. Super clients get all of the information from a server, not just the information that they need. However, this can cause a lot of unnecessary traffic. Nav2 has a lot of nodes so if they are all being launched as super client then that is not good news. Whenever you are launching a node from the terminal either first set `export ROS_SUPER_CLIENT=False` or prefix your command with `ROS_SUPER_CLIENT=False <your run or launch command>`. For example `ROS_SUPER_CLIENT=False ros2 run teleop_twist_keyboard teleop_twist_keyboard`. The first option will change that terminal to no longer act like a super client, the second will not change the terminal but it will specifically run that command with super client as false. If you launch using the service then it should not be a problem because when you source the setup.bash it will detect it as not a terminal and leave super client as false.

Please let me know if this helps, and which of this info you don't think is captured well in the documentation. I think there is definitely some improvements that can be made based on this discussion.

PrakritTyagi, 2024-12-25:
Thank you for your reply. This discussion has been really helpful. Sorry for my sentence formation, its probably confusing and so Ill explain my setup and situation below again in a better way. 

First I have taken your suggestion of creating a Ubuntu service for Navigation and localization. It was really helpfull. Do you have any suggestions for my implementation? I used a timer and service.

> my_tb4.timer

```
[Unit]
Description=Delay launching my_tb4.service by 180 seconds

[Timer]
OnBootSec=180s
Unit=my_tb4.service

[Install]
WantedBy=multi-user.target```
my_tb4.service
```

> my_tb4.service

```
[Unit]
Description=My TurtleBot4 custom launch

After=turtlebot4.service
Requires=turtlebot4.service

[Service]
Type=simple
User=ubuntu

# Source BOTH the turtlebot4 setup and your custom workspace
ExecStart=/usr/bin/bash -c " \
  source /etc/turtlebot4/setup.bash && \
  source /home/ubuntu/tb4_ros2_ws/install/setup.bash && \
  ros2 launch my_tb4 my_turtlebot4_startup.launch.py \
"

Restart=on-failure

[Install]
```
-----------------------------------------------------------------------------------------------------------------------------------
### Moving on about my issues with discovery when running ROS 2 nodes and services.

Current Setup.

- Four TB4 (2 lite 2 standard) with onboard discovery server running on raspi (NO interconnection between TB4s).
- On the PC I export the discovery servers using only this command `export ROS_DISCOVERY_SERVER="192.168.0.106:11811;192.168.0.41:11811;192.168.0.151:11811;192.168.0.112:11811;"` in terminals when I want to run some nodes or launch files. 
- ROS_SUPER_CLIENT is not exported, assuming its automatically false. 
- /etc/turtlebot4_discovery/setup.bash is also not sourced in the bashrc. Checked!
- **Test**: I created a launch file that runs four nodes to send goal location to each turtlebots. Launch file given below.
```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='turtlebot4_python_tutorials',
            executable='nav_to_pose1',
            name='nav_to_pose1',
            namespace='/robot1',
            output='screen',
        ),
        Node(
            package='turtlebot4_python_tutorials',
            executable='nav_to_pose2',
            name='nav_to_pose2',
            namespace='/robot2',
            output='screen',
        ),
        Node(
            package='turtlebot4_python_tutorials',
            executable='nav_to_pose3',
            name='nav_to_pose3',
            namespace='/robot3',
            output='screen',
        ),
        Node(
            package='turtlebot4_python_tutorials',
            executable='nav_to_pose4',
            name='nav_to_pose4',
            namespace='/robot4',
            output='screen',
        ),
    ])

```
- nav_to_pose1 python script looks like this (Other pythons scripts just have different goal locations)
```
import rclpy

from turtlebot4_navigation.turtlebot4_navigator import TurtleBot4Directions, TurtleBot4Navigator

def main():
    rclpy.init()

    navigator = TurtleBot4Navigator()

    # Start on dock
    # if not navigator.getDockedStatus():
    #     navigator.info('Docking before intialising pose')
    #     navigator.dock()

    # Set initial pose
    # initial_pose = navigator.getPoseStamped([0.0, 0.0], TurtleBot4Directions.NORTH)
    # navigator.setInitialPose(initial_pose)

    # Wait for Nav2
    # navigator.waitUntilNav2Active()

    # Set goal poses
    goal_pose = navigator.getPoseStamped([0.0, 2.108], TurtleBot4Directions.NORTH)

    # Undock
    # navigator.undock()

    # Go to each goal pose
    navigator.startToPose(goal_pose)

    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

### OUTCOME: 
- Nothing happens when I launch the above launch file. 
- What worked for me was this bash script for running the above python nav to goal scripts. This is what I meant by separating discovery servers.
```
#!/bin/bash

# Command for robot1
(
    export ROS_DISCOVERY_SERVER="192.168.0.106:11811;"
    ros2 run turtlebot4_python_tutorials nav_to_pose1 --ros-args -r __ns:=/robot1
) &

# Command for robot2
(
    export ROS_DISCOVERY_SERVER=";192.168.0.41:11811;"
    ros2 run turtlebot4_python_tutorials nav_to_pose2 --ros-args -r __ns:=/robot2
) &

# Command for robot3
(
    export ROS_DISCOVERY_SERVER=";;192.168.0.151:11811;"
    ros2 run turtlebot4_python_tutorials nav_to_pose3 --ros-args -r __ns:=/robot3
) &

# Command for robot4
(
    export ROS_DISCOVERY_SERVER=";;;192.168.0.112:11811;"
    ros2 run turtlebot4_python_tutorials nav_to_pose4 --ros-args -r __ns:=/robot4
) &

# Wait for all background processes to finish
wait

echo "All commands completed."

```

### Remarks

- Is my launch file approach wrong?
- What is the correct way to interact with mutiple TB4 in the case of sending simulatenous goals/ waypoint following/ cooperative task etc

hilary-luo, 2024-12-30:
> Do you have any suggestions for my implementation? I used a timer and service.

Is there a specific reason that you created the timer? You don't need to have a timer to trigger the service. Having created the service and having it enabled, it will automatically trigger according to the `After` condition.

I would also suggest adding the `RestartSec=3` parameter alongside `Restart=on-failure` parameter in `my_tb4.service` to add a slight delay on restart to give better confidence that the ROS components fully shutdown before they are restarted.

> Moving on about my issues with discovery when running ROS 2 nodes and services.

You are correct that super client will be false by default if you are not sourcing the turtlebot setup.bash, and you are not exporting it manually. It is generally good practice to check to ensure that it wasn't exported unintentionally. You can check it by running one of the options below:
```bash
printenv | grep ROS  # Will display all environment variables that have been exported and include the text ROS, case sensitive
echo $ROS_SUPER_CLIENT  # Will directly display the contents of the ROS_SUPER_CLIENT variable if it has been exported
```

> Is my launch file approach wrong?

Your launch file looks good to me, I am not sure why it is performing differently than your script although it does suggest that it could be an issue with the server IP's not being communicated properly in that terminal or that they are not subscribing properly. Can you share the full terminal output from after you launch it and then wait for a minute? In addition, after you run the launch file, can you check that the nodes were launched and discovered successfully (run `ros2 node list` in a new terminal with the correct discovery server settings) and then if present, check that the nodes are subscribing to the correct topics (run `ros2 node info </namespace/node_name>` where you replace `</namespace/node_name>` with the fully qualified name of the node that you are inspecting). If the node does not show up in node list it does not indicate whether it is a launch problem or a networking issue though. We can go from there in terms of troubleshooting where the issue is.

> What is the correct way to interact with mutiple TB4 in the case of sending simulatenous goals/ waypoint following/ cooperative task etc

What you have is a good place to start. When you start wanting to coordinate more complex cooperative tasks, then it will be beneficial to replace the turtlebot4 navigator with your own custom node. This node would be the collective director of all of the movements. In this node you would create your 4 instances of `BasicNavigator` (the node that turtlebot4 navigator extends), and you would directly call the Basic Navigator functions similar to how Turtlebot4Navigator calls them (see the [turtlebot4 navigator source code](https://github.com/turtlebot/turtlebot4/blob/humble/turtlebot4_navigation/turtlebot4_navigation/turtlebot4_navigator.py)). In this way your one node will have detailed feedback on each robot and be able to react and coordinate the robots. However, it is important to remember that all of this functionality is in one thread and to not block the thread. 

If you got to a point where the system was more complex than a single thread could handle then you could break some of it into separate nodes and use topics or services to pass information/commands between the nodes or you could create multiple threads out of one node. 

It is important to understand that the `Turtlebot4Navigator` is an abstraction of `BasicNavigator`. `Turtlebot4Navigator` offers a narrow set of functionality that is simple to implement in a script. `BasicNavigator` offers more functionality but is slightly more complex to implement. However, `BasicNavigator` is still an abstraction of the full Nav2 package that exists to make the Nav2 functionality available in a library format. At its core, Nav2 operates using the normal ROS 2 actions and services alongside the topics. I mention this because it can be helpful to know that how you use these simplified navigators is not how you normally interact with other ROS 2 nodes. Generally to trigger something to happen in a different node, it involves publishing to a topic, calling a service or using an action, not calling a member function.

closed 2025-01-20 by hilary-luo