turtlebot/turtlebot4 #499: Network Issue

Tauhid-Tanjim13, 2024-11-20:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

Package: ros-humble-turtlebot4-description
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 29045
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: arm64
Version: 1.0.5-1jammy.20240830.231101
Depends: ros-humble-irobot-create-description, ros-humble-joint-state-publisher, ros-humble-robot-state-publisher, ros-humble-urdf, ros-humble-ros-workspace
Description: TurtleBot4 Description package

### Type of issue

Networking

### Expected behaviour

Lights:
MTR: On
COMM: On
WIFI: On
BATT: On
POWER: On

### Actual behaviour

Instead, we observed:
Lights:
MTR: On
COMM: Off
WIFI: On
BATT: Off
POWER: On
Topics available on ros2 topic list:
/parameter_events  
/rosout  

When the robot is initially powered on, the light ring of the Create3 turns red 
[image]

After a short while, the light ring changes to another state
[image]

### Error messages

```bash
Of course, the topics have not been published properly.

Also, I got the error message from the create3 webserver's log. 
Nov 19 22:00:28 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/aos_misc.c:754 errno:No such file or directory) Failed to stat file /opt/irobot/persistent/opt/irobot/data/bbk/user_wpa_supplicant.conf
Nov 19 22:00:28 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/aos_misc.c:801 errno:No such file or directory) Failed to open file: /opt/irobot/persistent/opt/irobot/data/bbk/user_wpa_supplicant.conf
Nov 19 22:00:30 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/aos_kvs_helper.c:51 errno:No such file or directory) Failed to stat file network_password
Nov 19 22:00:40 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/aos_misc.c:8 errno:No such file or directory) Failed to retrieve key "robot.disc_reconnect_sleep_sec"
Nov 19 22:00:41 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/aos_misc.c:8 errno:No such file or directory) MODULE : network_manager CMD (sudo network_dhcp_arp_control.sh disable)
Nov 19 22:00:43 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/mqtt_client.c:785 errno:Resource temporarily unavailable) firr_transfer failed
Nov 19 22:00:47 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/discovery.c:407) failed to get cap from tspirme
Nov 19 22:01:11 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/mqtt_client.c:785) firr_transfer failed
Nov 19 22:04:44 iRobot-03B95E7A32904DDABE12FFE2D1690859 daemon.err [1467]: [ERROR] (src/provisioning_sm.c:637) Error publishing "disconnect_local" firr request.
```

### To Reproduce

1. Place the TB4 on the charging dock.
2. Wait
3. And ssh into the robot. (Not sure how to reproduce)

### Other notes

While I have resolved the problem a few times by following the instructions in the TurtleBot4 documentation (mostly simple discovery mode using rmw_fastrtps), the issue persists intermittently. We have nine robots, and not all of them are having this issue. 

Additionally, we failed to connect the remote PC to control the TurtleBot4 or echo ROS topics. As a workaround, we run all the programs directly on the Raspberry Pi (RPI) inside the robot, like map generation and SLAM. Even after resolving the connectivity issue using a simple Discovery Server setup, the problem reoccurs unpredictably.

Tauhid-Tanjim13, 2024-11-24:
I have flashed the Raspberry Pi again. Thank you. But I'll try my laptop to configure it. For some reason, it does not identify the topics—also, nine robots run under one institutional Wi-Fi.

smatarCPR, 2024-11-28:
Hello @Tauhid-Tanjim13 
Thank you for reaching out on the Turtlebot 4 Github page. I have a couple of questions regarding the setup of the Turtlebot:

1. Is the Turtlebot still in AP mode or have you been able to connect the Turtlebot 4 to wi-fi? 
2. Have you attempted to setup Simple Discovery or Discovery Server configuration for the Turtlebot 4? If using a Discovery Server, be sure to add a namespace to the ROS environment
3. What is the status of the status lights on the top plate? Is the COMM light green?
4. If you SSH into the Raspberry Pi are you able to view the ROS topics and echo them?

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-01-27:
Hello @Tauhid-Tanjim13 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif