turtlebot/turtlebot4 #501: Nav2 does not work on Jazzy

aharshac, 2024-11-25:
### Robot Model

Turtlebot4 Standard

### ROS distro

Select One

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

dpkg -l ros-jazzy-turtlebot4* | grep ^ii
> ii  ros-jazzy-turtlebot4-base        2.0.1-2noble.20241025.080249 arm64        Turtlebot4 Base Node
> ii  ros-jazzy-turtlebot4-bringup     2.0.1-2noble.20241118.200906 arm64        Turtlebot4 Robot Bringup
> ii  ros-jazzy-turtlebot4-description 2.0.1-1noble.20241118.200242 arm64        Turtlebot4 Description package
> ii  ros-jazzy-turtlebot4-diagnostics 2.0.1-2noble.20241025.080252 arm64        Turtlebot4 Diagnostics
> ii  ros-jazzy-turtlebot4-msgs        2.0.1-1noble.20240926.010643 arm64        Turtlebot4 Messages
> ii  ros-jazzy-turtlebot4-navigation  2.0.1-1noble.20241118.200018 arm64        Turtlebot4 Navigation
> ii  ros-jazzy-turtlebot4-node        2.0.1-1noble.20241018.082137 arm64        Turtlebot4 Node
> ii  ros-jazzy-turtlebot4-robot       2.0.1-2noble.20241118.201548 arm64        Turtlebot4 Robot Metapackage
> ii  ros-jazzy-turtlebot4-setup       2.0.2-1noble.20241025.080230 arm64        Turtlebot4 setup scripts
> ii  ros-jazzy-turtlebot4-tests       2.0.1-2noble.20241025.080005 arm64        Turtlebot4 System Tests

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

1. Robot description is published on `/robot_description`
2. Nav2 is initialized and robot shall move to goal

### Actual behaviour

The same sequence was working on Humble but now does not on jazzy:

1. Robot description is published on `/robot_description`, but empty (also seen on Humble)
2. Nav2 is not initialized
3. SSH connections to RPi are terminated and reconnected after a moment
4. Create 3 and RPu do not publish tf
5. nav2 local and global costmaps are not published

### Error messages

```bash
Logs to follow.
```

### To Reproduce

1. rviz2 -d <custom layout>
2. ros2 launch turtlebot4_navigation slam.launch.py
3. ros2 launch turtlebot4_navigation nav2.launch.py

aharshac, 2024-11-25:
## 1. Turtlebot 4 Intialized
[image]

## 2. slam_toolbox launched
[image]
[image]
Robot visualization cannot be seen.

aharshac, 2024-11-25:
## 2. slam_toolbox launched
[image]
[image]
Robot visualization cannot be seen.

Nodes:
/SlamToolboxPlugin
/SlamToolboxStateUpdateNode
/analyzers
/create3_repub
/joint_state_publisher
/joy_linux_node
/launch_ros_1338
/launch_ros_192027
/oakd
/oakd_container
/robot_state_publisher
/rplidar_composition
/rqt_gui_py_node_190567
/rviz
/rviz_navigation_dialog_action_client
/slam_toolbox
/teleop_twist_joy_node
/transform_listener_impl_581a8b2ebe10
/transform_listener_impl_581a8c196df0
/transform_listener_impl_624c17b81990
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node

aharshac, 2024-11-25:
## 3. nav2 started
**Nodes:**
/SlamToolboxPlugin
/SlamToolboxStateUpdateNode
/behavior_server
/bt_navigator
/collision_monitor
/controller_server
/docking_server
/global_costmap/global_costmap
/joint_state_publisher
/launch_ros_1338
/lifecycle_manager_navigation
/local_costmap/local_costmap
/oakd
/oakd_container
/planner_server
/robot_state_publisher
/rqt_gui_py_node_190567
/rviz
/rviz_navigation_dialog_action_client
/smoother_server
/teleop_twist_joy_node
/transform_listener_impl_57a9e45c4340
/transform_listener_impl_581a8b2ebe10
/transform_listener_impl_581a8c196df0
/transform_listener_impl_59c442156410
/turtlebot4_diagnostics
/turtlebot4_node
/velocity_smoother
/waypoint_follower

**Topics:**
/battery_state
/behavior_server/transition_event
/bt_navigator/transition_event
/clicked_point
/clock
/cmd_vel
/cmd_vel_nav
/collision_monitor/transition_event
/controller_server/transition_event
/diagnostics
/diagnostics_agg
/dock_status
/docking_server/transition_event
/downsampled_costmap
/downsampled_costmap_updates
/function_calls
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_raw_updates
/global_costmap/costmap_updates
/global_costmap/global_costmap/transition_event
/global_costmap/published_footprint
/global_costmap/voxel_marked_cloud
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/initialpose
/interface_buttons
/ip
/joint_states
/joy
/local_costmap/clearing_endpoints
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_raw_updates
/local_costmap/costmap_updates
/local_costmap/footprint
/local_costmap/local_costmap/transition_event
/local_costmap/published_footprint
/local_costmap/static_layer
/local_costmap/static_layer_raw
/local_costmap/static_layer_raw_updates
/local_costmap/static_layer_updates
/local_costmap/voxel_grid
/local_costmap/voxel_layer
/local_costmap/voxel_layer_raw
/local_costmap/voxel_layer_raw_updates
/local_costmap/voxel_layer_updates
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_updates
/mobile_base/sensors/bumper_pointcloud
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/oakd/rgb/preview/image_raw/zstd
/odom
/optimal_trajectory
/parameter_events
/particle_cloud
/plan
/plan_smoothed
/planner_server/transition_event
/robot_description
/rosout
/scan
/smoother_server/transition_event
/speed_limit
/tf
/tf_static
/trajectories
/transformed_global_plan
/velocity_smoother/transition_event
/waypoint_follower/transition_event
/waypoints
/wheel_status

smatarCPR, 2024-11-28:
Hello @aharshac ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear you are encountering issues with the Turtlebot 4. Would it be possible to provide the following details:

1. What version of the Create3 firmware version are you running? This can be found Create3 Webserver homepage or the Update tab - https://iroboteducation.github.io/create3_docs/webserver/home/ https://iroboteducation.github.io/create3_docs/webserver/update/
2. Double check the Create3's connect tab and ensure that no Wi-Fi connection is being established on the Create3, this may affect the Discovery server.
3. Are you running on NAV on the Raspberry Pi or the offboard computer (your computer)?
4. Are you running on a 2.4 GHz or 5.0 GHz network?

Looking forward to hearing from you soon.
Best Regards,
Saif

aharshac, 2024-11-29:
Hi @smatarCPR 

I solved the issues:
1. These were caused by bandwidth throttling by the WiFi network or due to Fast DDS.
Solved by using Cyclone DDS and  [zenoh-plugin-ros2dds](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds) between RPi and PC.
>   1. Robot description is published on /robot_description, but empty (also seen on Humble)
>   2. Nav2 is not initialized
>   3. SSH connections to RPi are terminated and reconnected after a moment
>   4. Create 3 and RPi do not publish tf
2. nav2 costmaps not being published was because of `use_sim_time` being set true by the default config of turtlebot4_navigation.
Solved by copying the default nav2 config and setting `use_sim_time: false`

aharshac, 2024-11-29:
## Zenoh config

### 1. RPi
```json
{
    "mode": "peer",
    "listen": { "endpoints": ["tcp/0.0.0.0:7447"] },
    "plugins": {
        "ros2dds": {
            "namespace": "/bot1",
            // "deny": {
            //     "publishers": [".*/_internal/.*"],
            //     "subscribers": [".*/_internal/.*"],
            //     "service_servers": [".*/_internal/.*"],
            //     "service_clients": [".*/_internal/.*"],
            //     "action_servers": [".*/_internal/.*"],
            //     "action_clients": [".*/_internal/.*"]
            // },
            "allow": {
                "publishers": [
                              ".*/battery_state",
                              // ".*/cliff_intensity",
                              // ".*/cmd_audio",
                              // ".*/cmd_lightring",
                              ".*/cmd_vel",
                              ".*/dock_status",
                              // ".*/hazard_detection",
                              ".*/imu",
                              // ".*/interface_buttons",
                              ".*/ir_intensity",
                              ".*/ir_opcode",
                              ".*/joint_states",
                              // ".*/kidnap_status",
                              // ".*/mobility_monitor/transition_event",
                              // ".*/mouse",
                              ".*/oakd/.*",
                              ".*/odom",
                              ".*/robot_description",
                              ".*/robot_state/transition_event",
                              ".*/scan",
                              // ".*/slip_status",
                              ".*/static_transform/transition_event",
                              // ".*/stop_status",
                              ".*/tf",
                              ".*/tf_static",
                              ".*/wheel_status",
                              // ".*/wheel_ticks",
                              // ".*/wheel_vels",
                              // "/clock",
                              // "/parameter_events",
                              // "/rosout",
                              // "/set_elevator",
                            ],
                "subscribers": [
                              ".*/battery_state",
                              // ".*/cliff_intensity",
                              // ".*/cmd_audio",
                              // ".*/cmd_lightring",
                              ".*/cmd_vel",
                              ".*/dock_status",
                              // ".*/hazard_detection",
                              ".*/imu",
                              // ".*/interface_buttons",
                              ".*/ir_intensity",
                              ".*/ir_opcode",
                              ".*/joint_states",
                              // ".*/kidnap_status",
                              // ".*/mobility_monitor/transition_event",
                              // ".*/mouse",
                              ".*/oakd/.*",
                              ".*/odom",
                              ".*/robot_description",
                              ".*/robot_state/transition_event",
                              ".*/scan",
                              // ".*/slip_status",
                              ".*/static_transform/transition_event",
                              // ".*/stop_status",
                              ".*/tf",
                              ".*/tf_static",
                              ".*/wheel_status",
                              // ".*/wheel_ticks",
                              // ".*/wheel_vels",
                              // "/clock",
                              // "/parameter_events",
                              // "/rosout",
                              // "/set_elevator",
                             ],
                "service_servers": [".*/start_motor", ".*/stop_motor", ".*/reset_pose"],
                "service_clients": [".*/start_motor", ".*/stop_motor", ".*/reset_pose"],
                "action_servers": [".*/dock", ".*/undock"],
                "action_clients": [".*/dock", ".*/undock"],
            },
            "pub_max_frequencies": [
                ".*/image_raw=10",
                ".*/camera_info=10"
            ]
        },
        "rest": { "http_port": 8000 }
    }
}
```

### 2. PC
Replace `"listen": { "endpoints": ["tcp/0.0.0.0:7447"] },` with `"connect": { "endpoints": ["tcp/<RPi IP address>:7447"] },`

aaronchongth, 2024-12-05:
I found that peer mode would still sometimes be flaky, and starting a zenoh router on the host PC, before starting the bridges' with mode `client` makes it repeatable

aharshac, 2024-12-05:
@aaronchongth Thanks for the tip! 

Do you mean something like this?
### PC
1. `zenoh-bridge-ros2dds -m router`
2. `zenoh-bridge-ros2dds -c ~/zenoh_pc.json5` <-- mode changed to *client*

### RPi
`zenoh-bridge-ros2dds -c ~/zenoh_rpi.json5` <-- mode changed to *client*

aaronchongth, 2024-12-05:
@aharshac, it would be using the `zenohd` command to start a router
you can check out https://zenoh.io/docs/getting-started/installation/

closed 2025-04-14 by smatarCPR