turtlebot/turtlebot4 #502: building empty workspace on pi fails in iron image

mdxtinkernick, 2024-11-30:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

testing now with image as provide for jazzy on pi 2.0.1

### Type of issue

Other

### Expected behaviour

create an empty workspace on pi and successfully run colcon build

### Actual behaviour

colcon build fails, and rosdep as well ModuleNotFoundError: No module named 'catkin_pkg.package'

### Error messages

```bash
ModuleNotFoundError: No module named 'catkin_pkg.package'
```

### To Reproduce

on pi

mkdir ros_ws
cd ros_ws
mkdir src
colcon build

### Other notes

tried

sudo apt install python3-catkin-pkg

python3-catkin-pkg is already the newest version (1.0.0-100)

googling this turned up some mentions about condo environments - but don't think the pi image is using that, so don't know how to fix this

mdxtinkernick, 2024-11-30:
have built an image from ubuntu server 2404 then running the turtlebot4_setup.sh on it. This runs colcon build without the error, and rosdep works as well

smatarCPR, 2024-12-20:
Hello @mdxtinkernick 
Thank you for reaching out on the Turtlebot 4 Github. Apologies on the late reply for this, but it seems you were able to resolve the matter. I just wanted to check in and see if there was anything else we could help with. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-01-20:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2025-01-20 by smatarCPR

RA2004, 2025-07-01:
Hello @mdxtinkernick
I am having the same issue with my turtlebot4 running ubuntu 24.04. Would you be able walk me through the steps you took and provide any links you have referenced (such as where to find the imager or any other tutorial). 

Much thanks!