turtlebot/turtlebot4 #503: Create3 misconfiured? no movement

jsantore, 2024-12-03:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ros-humble-turtlebot4-description/jammy,now 1.0.5-1jammy.20240830.212943 amd64 [installed,automatic]
ros-humble-turtlebot4-desktop/jammy,now 1.0.0-1jammy.20240830.234728 amd64 [installed]
ros-humble-turtlebot4-viz/jammy,now 1.0.0-1jammy.20240830.233602 amd64 [installed,automatic]

### Type of issue

Other

### Expected behaviour

I've followed the directions for setting the robot up with discovery server networking, but I think I must have missed a step because nothing that involves the create3 seems to work.

in particular, I've tried the teleop_twist program to try to drive the robot - when using it I expect the robot to move

### Actual behaviour

no movement occurs

### Error messages

```bash
no error messages

here is the create3 about/system info page on the create3 webserver
System Info

>>>>>>> system IP address
unknown

>>>>>>> system version
H.2.6

>>>>>>> system sku
RCi3099

>>>>>>> system battery
Percentage:10
Capacity(mAh):2113

>>>>>>> system properties
   PRODUCT         : create3
   SERIAL          : e17628
   CLEAN MODE      : normal
   CONNS SVC MODE  : normal
   AP RADIO        : wlan1/g/6/[SHORT-GI-20][GF]/Create-D5A1/US
   AP IP ADDR      : 192.168.10.1
   CL IP           : dhcp/#/#/#
   CL DNS          : #/#/#
   CL DHCPCD T/O   : 3
   CL REG DOMAIN   : #
   ETH0 IP         : static/192.168.186.2/192.168.186.3
   USB0 IP         : 192.168.186.2/192.168.187.2/192.168.186.3
   SYSTEM_ACCESS   : locked
   DEVELOPMENT MODE: disabled
   RADIO_LIST      : wlan0 wlan1
   HOSTNAME        : iRobot-D5A10254D4C24087986629BF59596BC5
   ROBOT ID        : D5A10254D4C24087986629BF59596BC5
   BOOT COUNT      : 26
   LANGUAGE PACK   : unknown
   EXTRA LANGUAGE  : not_installed

>>>>>>> system date
Tue Dec  3 16:01:39 UTC 2024

>>>>>>> system uptime
 16:01:39 up 13 min,  load average: 1.50, 1.22, 0.74

>>>>>>> system memory
              total        used        free      shared  buff/cache   available
Mem:          59088       30360        1312         176       27416       26876
Swap:             0           0           0

>>>>>>> system ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:128 errors:0 dropped:0 overruns:0 frame:0
          TX packets:128 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7766 (7.5 KiB)  TX bytes:7766 (7.5 KiB)

usb0      Link encap:Ethernet  HWaddr F2:C2:78:33:45:C0  
          inet addr:192.168.186.2  Bcast:192.168.186.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7695 errors:0 dropped:5 overruns:0 frame:0
          TX packets:97553 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:934226 (912.3 KiB)  TX bytes:32697297 (31.1 MiB)

>>>>>>> system hciconfig

Here is the ros2 topic list from the pi on the turtlebot4
ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status

and here is the ros2 topic list from the computer I'm using 
~/Robots$ ros2 topic list
/battery_state
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
```

### To Reproduce

run 
`ros2 run teleop_twist_keyboard teleop_twist_keyboard`

type k or i

nothing happens with the robot

### Other notes

the 
`ros2 topic echo /battery_state
`
never returns anything though the /ip topic does show output

`ros2 topic echo /ip`
data: 192.168.1.6
---
data: 192.168.1.6
---

This might be user error (I'd love it to be) if there is anything else helpful to diagnose this please let me know 
thanks

smatarCPR, 2024-12-31:
Hello @jsantore ,
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry for the late on this.

I have the following questions/recommendations regarding your current setup:

- What is the status of the top LED lights on the Turtlebot 4?
- You mentioned running a discovery server setup, but I notice in the outputs you have provided that there is no namespace present. For a discovery server setup a namespace is required in order for the Create3 and Raspberry Pi to communicate. Add a namespace to the ROS configuration of the Turtlebot 4 Setup utility and see if the behaviour changes

Looking forward to hearing from you soon.
Best Regards,
Saif

jsantore, 2024-12-31:
Thanks very much - I'll be in the office with the create on Thursday, the day after New Years and will try this out (and get back to you about the LEDs)

jsantore, 2025-01-02:
[image]
Here is the status lights for the Turtlebot in question after about 4 hours of charging. I'll work with namespaces when I next get a chance. I spent more time charging than anything else today.

jsantore, 2025-01-07:
adding the namespace allowed the tutorial the run and topics to publish and be viewed. (also accurate battery data is now showed on the display) Now everything except the 
`ros2 run teleop_twist_keyboard teleop_twist_keyboard

`
seems to work, the teleop_twist still doesn't work - any ideas on that? sending a custom message using 

ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0" 
  (with values ) will move the robot
  
  thanks for helping me get this far.

jsantore, 2025-01-10:
ok, So it turns out the reason teleop_twist_keyboard doesn't work is because it doesn't use namespaces. After downloading teleop twist and editing it locally to use the namespace, teleop_twist_keyboard works now, so the original question is now solved.

I have a couple of follow ups, chief among them, I'm finding that something is crashing on the robot fairly often. I need to use the restart from robot upstart at least a couple of times a day - is there a fix for this?

should I start a new issue for that?

smatarCPR, 2025-01-14:
Hello @jsantore 
Glad to hear you were able to resolve the matter. In terms of the new issue, for the sake of better tracking please open a new ticket for the crashing you are seeing. Please include the Turtlebot 4 `logs journalctl -b > turtlebot4_log.txt` as well as he Create3 logs https://iroboteducation.github.io/create3_docs/webserver/logs/

Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2025-01-14 by smatarCPR