turtlebot/turtlebot4 #504: [Notice] Nav2 Migrated to TwistStamped to replace Twist for cmd_vel Topics

SteveMacenski, 2024-12-03:
### Describe the the feature you would like

Hi,

As a key community project using Nav2, I wanted to drop you a note that I am in the process of changing the default cmd_vel topic type from Twist to TwistStamped in efforts of enabling a broader set of applications and safety constraints. This has been an on-going discussion over the last 18 months and now has the technology stack in GZ, ROS Control, and Nav2 to enable it.

To migrate, your robot and simulation should soon accept a TwistStamped message rather than a Twist message. We have a couple of provided utilities that could make this easier for you, by switching to use a Twist Publisher and Twist Subscriber [1-2]. This is what is now used in all Nav2 nodes to either publish or subscribe to a Twist or TwistStamped message.

If you would like to continue to use a Twist, you can set enable_stamped_cmd_vel to false in all nodes in your configuration that use Twist Publisher/Subscriber objects, but this will be shortly switched to being default true.

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_publisher.hpp
[2] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_subscriber.hpp

Nav2 migration:
- https://github.com/ros-navigation/navigation2/pull/4779
- https://github.com/ros-navigation/docs.nav2.org/pull/614
- https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/pull/16

### Motivation and impact

Enable twist stamped for default use

hilary-luo, 2024-12-30:
@SteveMacenski Thanks for letting us know. Can you confirm that this is only affecting Rolling and future distros (K+)?

SteveMacenski, 2025-01-02:
Yes indeed!