turtlebot/turtlebot4 #505: Navigation not working correctly on already mapped file [Humble - Tb4 Standard]

HevCuesta, 2024-12-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.5-1jammy.20240831.002728

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Normal navigation after setting initial pose, able to set goal poses for the robot to navigate through

### Actual behaviour

Launching both localization and nav2 with the included packages (I have also tried building from source with no success), ends up looping with the following message on nav2:

[controller_server-1] [INFO] [1733335956.823897579] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

Attached are the tf tree and topic list
[frames_2024-12-04_19.12.11.pdf](https://github.com/user-attachments/files/18012516/frames_2024-12-04_19.12.11.pdf)
Topic list:
```
daniel@ubuntu-ros2 ~> ros2 topic list
/amcl/transition_event
/amcl_pose
/battery_state
/behavior_server/transition_event
/bond
/bt_navigator/transition_event
/cmd_vel
/cmd_vel_nav
/cmd_vel_teleop
/controller_server/transition_event
/cost_cloud
/diagnostics
/dock_status
/evaluation
/function_calls
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_updates
/global_costmap/footprint
/global_costmap/global_costmap/transition_event
/global_costmap/published_footprint
/goal_pose
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/initialpose
[... 239 lines cut ...]
[behavior_server-4] [INFO] [1733335760.401707984] [behavior_server]: Configuring wait
[lifecycle_manager-8] [INFO] [1733335760.512896599] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-5] [INFO] [1733335760.513261393] [bt_navigator]: Configuring
[lifecycle_manager-8] [INFO] [1733335761.389880767] [lifecycle_manager_navigation]: Configuring waypoint_follower
[waypoint_follower-6] [INFO] [1733335761.390109827] [waypoint_follower]: Configuring
[waypoint_follower-6] [INFO] [1733335761.535080400] [waypoint_follower]: Created waypoint_task_executor : wait_at_waypoint of type nav2_waypoint_follower::WaitAtWaypoint
[lifecycle_manager-8] [INFO] [1733335761.535881983] [lifecycle_manager_navigation]: Configuring velocity_smoother
[velocity_smoother-7] [INFO] [1733335761.536098619] [velocity_smoother]: Configuring velocity smoother
[lifecycle_manager-8] [INFO] [1733335761.557688842] [lifecycle_manager_navigation]: Activating controller_server
[controller_server-1] [INFO] [1733335761.557897704] [controller_server]: Activating
[controller_server-1] [INFO] [1733335761.557930485] [local_costmap.local_costmap]: Activating
[controller_server-1] [INFO] [1733335761.557943279] [local_costmap.local_costmap]: Checking transform
[controller_server-1] [INFO] [1733335761.557963146] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [INFO] [1733335762.058180757] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [INFO] [1733335762.558033464] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [INFO] [1733335763.058050439] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [INFO] [1733335763.558052866] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
...
```

### To Reproduce

1. `ros2 launch turtlebot4_navigation localization.launch.py map:=uem_floor1_clean.yaml`
2. `ros2 launch turtlebot4_navigation nav2.launch.py`
3. Launch rviz2 normally, also tried with the view_robot launcher

HevCuesta, 2024-12-05:
Forgot to add that adding initial pose does not help, I have tried with several parameters such as sim_time in case it is a time sync problem, as well as changing the DDS implementation

smatarCPR, 2024-12-20:
Hello @HevCuesta 
Thank you for reaching out on the Turtlebot 4 Github page. I have a few questions regarding your current setup:

- Reviewing your outputs it appears there is a time syncing issue there are messages regarding data having a timestamp earlier that all data in the queue. Are you able to resync the timing between the Create3, the Raspberry PI, and your computer. To do so follow the details here https://github.com/turtlebot/turtlebot4/issues/216
- It appears that in the last few lines it seems that the odom is unable pass its transform data, would it be possible to check that the Create3 topics are able to be echoed on the Raspberry Pi.
- I had a quick question regarding the network setup: you mention being on a simple discovery setup, do you have multiple frequencies setup? Is the Create3 on a 2.4 GHz and the Pi on a 5.0 GHz?
- Are you able to swap to a discovery server setup or is something preventing you from doing so?

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

HevCuesta, 2024-12-26:
Alright, will check out the solutions you suggested as soon as I get ahold of the robot, cheers.

HevCuesta, 2025-01-28:
Hello,

Sorry for the late response, I tried the suggested solutions but none of them ended up working. In the end the only fix that ended up working was doing a fresh install and switching to CycloneDDS.

Thanks for the help, cheers.