turtlebot/turtlebot4 #506: turtlebot's nodes killed when receiving cmd_vel from remote node

mwsatman, 2024-12-05:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

not sure which package is causing this issue

### Type of issue

Other

### Expected behaviour

The turtlebot4 moves when receiving cmd_vel from both 
1. Simple Twist publisher
and 
2. ros2 topic pub

ros2 topic pub -r 10 /tb4_01/cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.5"

### Actual behaviour

The turtlebot moves when the cmd_vel is sent via ros2 topic pub.
But, when the cmd_vel is sent via simple Twist Publisher the node in Turtlebot4 suddenly killed. 

If the robot upstart is restarted via turtlebot4-setup, the nodes and topics active again.
cmd_vel from ros2 topic pub moves the robot.
But still crashes when receive message from simple Twist Publisher.

### Error messages

```bash
The output when echoing the topic in the raspberry pi via ssh

1. For the ros2 topic pub --> The robot spins

ubuntu@ubuntu:~$ ros2 topic echo /tb4_01/cmd_vel

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Fast CDR exception deserializing message of type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_., at ./src/type_support_common.cpp:118'

with this new error message:

  ''Bad alloc' exception deserializing message of type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_., at ./src/type_support_common.cpp:123'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.5
---
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.5
---
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.5
---

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  ''Bad alloc' exception deserializing message of type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_., at ./src/type_support_common.cpp:123'

with this new error message:

  'Fast CDR exception deserializing message of type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_., at ./src/type_support_common.cpp:118'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

```

2. For the simple Twist Publisher
```
ubuntu@ubuntu:~$ ros2 topic echo /tb4_01/cmd_vel
Killed
```
```

### To Reproduce

1. Update Create3 firmware to H.2.6 and then factory reset
2. Install turtlebot4_lite_humble_1.0.0 on raspberry pi and `sudo apt update && sudo apt upgrade` to the latest version
3. Via turtlebot4-setup, set ROBOT_NAMESPACE into `tb4_01` and set  Discovery Server
4. Setup Discovery server setting on Remote PC (Jazzy)

5. Publish the command from the terminal on Remote PC --> The robot spins
ros2 topic pub -r 10 /tb4_01/cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.5"

6. Run the simple Twist Publisher (in other notes) --> The ros2 nodes in turtlebot4 are killed.

### Other notes

The script for the simple Twist Publisher
```
import rclpy
from rclpy.node import Node

from geometry_msgs.msg import Twist

class MinimalPublisher(Node):

    def __init__(self):
        super().__init__('minimal_publisher')
        self.publisher_ = self.create_publisher(Twist, '/tb4_01/cmd_vel', 10)
        timer_period = 0.5  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.i = 0

    def timer_callback(self):
        TBvel = Twist()
        TBvel.linear.x = 0.0
        TBvel.linear.y = 0.0
        TBvel.linear.z = 0.0
        TBvel.angular.x = 0.0
        TBvel.angular.y = 0.0
        TBvel.angular.z = 0.5

        self.publisher_.publish(TBvel)
        self.get_logger().info('Publishing: Twist data')
        self.i += 1

def main(args=None):
    rclpy.init(args=args)

    minimal_publisher = MinimalPublisher()

    rclpy.spin(minimal_publisher)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

mwsatman, 2024-12-05:
I put the simple Twist Publisher locally in the raspberry Pi, colcon build it and run locally. 
And the setup manages to move the turtlebot. 

So it seems the problem is only occurs when I am running the Publisher from the remote PC (Jazzy). 
Is there any advice on what the problem might be and how to solve it?

Additionally, due to the safety settings on the Remote PC, it is not possible to install Ubuntu 22 and Humble on it.

smatarCPR, 2024-12-11:
Hello @mwsatman 
Thank you for reaching out on the Turtlebot 4 Github page. Based on the detail provided I have the following information/recommendations:

- Cross distro communication (Humble to Jazzy) is not supported and can lead to various issues, particularly the one you are currently encountering.
- If you are unable to have the remote PC run ROS Humble, then the recommended action is to upgrade the Turtlebot to run ROS Jazzy. Instructions on how to upgrade to Jazzy can be found on our online documentation here - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

mwsatman, 2024-12-17:
Hi, 
Thanks for your comments.
After several testing and readings discussions online, I agree with you that the cross distro setup is the core problem.
I resolved this issue by using docker to run humble on my PC with multiple turtlebots in discovery mode. 
So far, it is working well.

Best regards, Widhi

closed 2024-12-17 by mwsatman