turtlebot/turtlebot4 #508: altering settings for rplidar

mdxtinkernick, 2024-12-06:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

Running Jazzy image 2.0.1

### Type of issue

LIDAR

### Expected behaviour

I want to edit the settings for the rplidar so it doesn't stop while in the dock - don't understand how to get it to use them when it boots. 

I have tried cloning the turtlebot4_robot package into a workspace on the pi. Editing the turtlebot4_bringup/launch/config/turtlebot4.yaml file and setting power_saver: False

Building the workspace and sourcing it in bashrc, and then rebooting, but the lidar still stops when it is put on the dock. 

Is this the right way to try and do it, but additional steps need to be taken, or am I going in totally the wrong direction

Thanks

### Actual behaviour

change does not stop the lidar stopping spinning when the robot docks

### To Reproduce

clone the package into a workspace and edit the file, add source command to bashrc and reboot the pi

mdxtinkernick, 2024-12-11:
in addition altering the parameter power_saver to false using the ros2 param set  successfully changes the parameter, but does not change the behaviour of the robot, the lidar still stops spinning once it placed on the charger

RustyCPR, 2025-01-27:
Closing due to inactivity

closed 2025-01-27 by RustyCPR